prusa3d/PrusaSlicer #1440: honeycomb infill layer inserted between solid infill layers

LenRemmerswaal, 2018-11-28:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

LenRemmerswaal, 2018-11-28:
**Version**  1,41.0+win64
**OS** W10 Home 1803
### Behavior
This is to report what seems like a bug to me.

A gcode was produced that extends a honeycomb infill between a local series of solid infill layers, thus producing an ugly top layer. Error occurred at layer 15, at 2,8 mm.  This was the first time I tried a honeycomb fillin.

Attached are layers 14, 15 and 16 and the zipped gcode. 

Layer 14:
[image]
Layer 15:
[image]
Layer 16:
[image]
Produced gcode and config file
[Failed_Slicing_Enclosure_Bottom_Corners_set2__0.2mm_hc_PET_MK3_7h07.zip](https://github.com/prusa3d/Slic3r/files/2626945/Failed_Slicing_Enclosure_Bottom_Corners_set2__0.2mm_hc_PET_MK3_7h07.zip)
STL file: It was split and some parts taken out to have 2 print jobs of manageable duration
[Prusa_Enclosure_Bottom_Corners.stl](https://www.thingiverse.com/download:4828239)

LenRemmerswaal, 2018-11-29:
More data.
I isolated the part that went wrong and sliced again. Error persists.
I tried the stoc Prusa FAST settings with rectangular grid infill. Error persists.
I rotated the part 90 degr. Error gone.
I reloaded my honeycomb settings (stock FAST 0.2 settings with only infill type changed): error gone.
Hope this helps.

supermerill, 2018-11-29:
Can you try with the latest release?

Can you export your plater where the setting makes it fail? (Plater->export to .3mf)

FidelCapo, 2018-12-04:
It seems to me that this issue is fixed in version 1.41.2. Can you confirm this? Thank you.

LenRemmerswaal, 2018-12-05:
Confirmed.
That was quick  :-)
Thanks

closed 2018-12-05 by LenRemmerswaal