prusa3d/PrusaSlicer #1442: Attempting to cut a multi part MMU model crashes Slic3r

TimothyGold, 2018-11-29:
### Version
Slic3r 1.41.2 - Prusa Edition

### Operating system type + version
Windows 10

### Behavior
Attempting to get a finished print with MMU2 is very challenging for me so in an effort to complete a print I finally thought I would continue a print by cutting the finished print off and continue up and glue the 2 together.  I have done this before when the MK3 suffered by layer shifting.  I tried several times to cut the multipart model but Slic3r always crashes and never cuts the model.   

#### STL/Config (.ZIP) where problem occurs
Model attempting to cut - 
https://www.thingiverse.com/thing:3090119

FidelCapo, 2018-12-13:
Thank you for report. I can reproduce this issue with Slic3r PE 1.41.2. I also test cutting this model in version 1.42.0-alpha that will be released soon and in this version cutting this object seems to me ok. Can you please retest this issue with new version 1.42.0-alpha when will it be released? Thank you.

bubnikv, 2019-02-05:
The crash now is fixed on master by swapping the triangulation library with a proven glu tesselator (written in 1994 in plain C by SGI). The fix will be part of the Slic3r 1.42.0-alpha6.

While the triangulator does not crash, if the input mesh is broken, the cut is not always triangulated. Namely, inverted parts are not triangulated.

Closing.

closed 2019-02-05 by bubnikv