prusa3d/PrusaSlicer #1443: Top layers cause stuck on Preparing Infill

nadavge, 2018-11-30:
### Version
1.41.1+win64

### Operating system type + version
Windows 10 Home 64-bit 
### Behavior
I decided to attempt to set the top layers count to 9999 and the infill to 100% in order to print miniatures. However, when slicing this seems to lead to program getting stuck preparing infill.

This had been reproduced when setting the infill from either 1% to 100%, and starting from 39 top layers (inclusive).

(This could be an expenonetially growing runtime bug, and therefore the exact top layers number is obviously not determinal).

#### STL/Config (.ZIP) where problem occurs
[bug_report.zip](https://github.com/prusa3d/Slic3r/files/2632964/bug_report.zip)

supermerill, 2018-12-03:
Not a good idea.
You should set the top & bottom layer to 1 and set infill ratio to 100%

acorderob, 2018-12-07:
I use 0 top, 999999 bottom, and 0% infill for miniatures and it seems to work.

rtyr, 2021-01-22:
Still happen with the PrusaSlicer 2.3.0

g3org3z, 2021-02-24:
> Still happen with the PrusaSlicer 2.3.0

I have experienced it in Prusa 2.3 as well.
[image]

closed 2026-09-29 by prusaslicer-bot[bot]