prusa3d/PrusaSlicer #1444: Still issue with 3D View and Preview

rikthefrog, 2018-11-30:
### Version
_Version of Slic3r used goes here_1.41.2+win64

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_Windows 10 see below of output System information

### Behavior
* _Describe the problem_
Issue is that scale of the screen is much to big see screenshot. No way to zoom out in both preview and 3D mode. Only the layers pane shows the model.
[image]

See layers pane
[image]

below is the top view in the preview pane
[image]

* _Steps needed to reproduce the problem_
Load STL from file

[trawler.zip](https://github.com/prusa3d/Slic3r/files/2635021/trawler.zip)

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
Want to be able to see the model in the preview pane
* _Actual Results_
Model is so zoomed out in the preview pane that only small parts are visible also in 3D Pane
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Slic3r Prusa Edition
Version: 1.41.2+win64
Build:   Slic3rPE-1.41.2+win64-full-201811221420
Operating System:    MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: 
Microsoft Windows [Version 10.0.17134.407]

Summary of my perl5 (revision 5 version 24 subversion 0) configuration:
   
  Platform:
    osname=MSWin32, osvers=10.0, archname=MSWin32-x64-multi-thread
    uname=''
    config_args='undef'
    hint=recommended, useposix=true, d_sigaction=undef
    useithreads=define, usemultiplicity=define
    use64bitint=define, use64bitall=undef, uselongdouble=undef
    usemymalloc=n, bincompat5005=undef
  Compiler:
    cc='cl', ccflags ='-nologo -GF -W3 -O1 -MD -Zi -DNDEBUG -GL -fp:precise -DWIN32 -D_CONSOLE -DNO_STRICT -DWIN64 -DCONSERVATIVE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_NONSTDC_NO_DEPRECATE  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DWIN32_NO_REGISTRY',
    optimize='-O1 -MD -Zi -DNDEBUG -GL -fp:precise',
    cppflags='-DWIN32'
    ccversion='18.00.40629', gccversion='', gccosandvers=''
    intsize=4, longsize=4, ptrsize=8, doublesize=8, byteorder=12345678, doublekind=3
    d_longlong=undef, longlongsize=8, d_longdbl=define, longdblsize=8, longdblkind=0
    ivtype='__int64', ivsize=8, nvtype='double', nvsize=8, Off_t='__int64', lseeksize=8
    alignbytes=8, prototype=define
  Linker and Libraries:
    ld='link', ldflags ='-nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02"'
    libpth=\lib
    libs=oldnames.lib kernel32.lib user32.lib gdi32.lib winspool.lib  comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib  netapi32.lib uuid.lib ws2_32.lib mpr.lib winmm.lib  version.lib odbc32.lib odbccp32.lib comctl32.lib msvcrt.lib
[... 377 lines cut ...]
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

rikthefrog, 2018-11-30:
[slic3r.zip](https://github.com/prusa3d/Slic3r/files/2635037/slic3r.zip)
this is the slic3r.ini file

rikthefrog, 2018-11-30:
[config.zip](https://github.com/prusa3d/Slic3r/files/2635041/config.zip)
plus config.ini

bubnikv, 2018-12-13:
@FidelCapo Does it still happen with the Slic3r 1.42 alpha?

FidelCapo, 2018-12-14:
@bubnikv Unfortunately, I've never been able to reproduce this issue even in version 1.41.2.

bubnikv, 2019-01-29:
I hope this issue was fixed in current Slic3r 1.42.0-alpha4. Closing.

closed 2019-01-29 by bubnikv