prusa3d/PrusaSlicer #1445: first layer of support for multi filament support oddly spaced

Mirarkitty, 2018-11-30:
### Version
Slic3r PE 1.41.2

### Operating system type + version
Linux

### Behavior
* Using Slic3r with MMU2 with PET and PVA+, contact z distance 0 to turn on MMU2 soluble support
* support base turns very stringy; not solid

[image]

* since PET and PVA+ have stickiness issues, this makes it 100% likely that it will unravel this when doing layer 2. 

I have no idea why the first layer turns out like this. It's useless and there's no option of making a solid layer. Pattern distance settings have no impact.

FidelCapo, 2018-12-04:
Can you please export and send me 3MF file with this issue? Thank you.

Mirarkitty, 2018-12-05:
http://www.mirar.org/graces.3mf

Mirarkitty, 2018-12-05:
[image]
Even with solid support (distance=0) it makes that weird first layer.

FidelCapo, 2018-12-06:
Spacing distance of soluble supports in first layer is hardcoded in slic3r and can not be changed by user. We plan to increase this value from 50% to 75% in future. I think #1165 is a very similar issue.

Mirarkitty, 2018-12-06:
Can you please make it configurable? I want it at 0 for a solid base, just like normal supports.

Kraligor, 2018-12-07:
> Can you please make it configurable? I want it at 0 for a solid base, just like normal supports.

This. Slic3r is full of options already, and users love it for exactly that fact. Randomly choosing settings to be non-configurable is just weird, especially when they seem to restrict Slic3r's versatility.

totalitarian, 2019-01-11:
Agreed, we need to be able to set solid supports. I end up having to use a raft as a workaround :(

bubnikv, 2021-02-24:
Fixed with fcb714cd241f0847522b9aac57da6b62acdd0152

Changed the 1st layer infill to rectilinear even for soluble materials.
New parameters raft_first_layer_density and raft_first_layer_expansion
to influence the 1st layer of raft or support.

closed 2021-02-24 by bubnikv