prusa3d/PrusaSlicer #1446: M221 set at 95 in start gcode and never changes again.  MMU2 Slic3r 1.41.2

tjcolejr, 2018-12-01:
### Version
1.41.2+win64

### Operating system type + version
Win 10

### Behavior

The start gcode for the MMU2 has the following line:

M221 S{if layer_height<0.075}100{else}95{endif}

So on a .2mm layer height, flow is set to 95% from the very first layer and stays that way throughout the print UNLESS you change it on the LCD and then it sticks because there is only one 221 command written by Slic3rPE 1.41.2 (yes, I checked).

I was under the impression but could very well be wrong that "Flow" was analogous to "Extrusion Multiplier" and should be set with each individual filament change based upon that specific filament's setting. Is this a bug in Slic3rPE?

I see NO code setting my multiplier for each filament. I do see the code for the linear advance factor, so it appears that the filament specific gcode is being written by Slic3r. 221 could be set there until this is sorted???

_Is this a new feature request?_ No 

#### STL/Config (.ZIP) where problem occurs

[Benchy.zip](https://github.com/prusa3d/Slic3r/files/2635427/Benchy.zip)

lukasmatena, 2018-12-01:
These are both intentional. The 95 % flow rate is used to compensate for overextrusion, there is a forum post about it at https://shop.prusa3d.com/forum/others-archive--f66/updated-slic3r-pe-over-extrusion-and-cooling-solve-t12949.html. The fine tuning is done by the change of flow rate in the printer custom g-code and we have found that 95 % works best for the MK3. The flow rate setting is also used for precise extruder calibration, which is why it is not affected by filament settings.

Extrusion multiplier in filament settings does not touch the flow rate override. Instead it generates all G1 codes with proper E/distance ratios. You can easily try that by generating a gcode, changing the filament extrusion multiplier, generating gcode again and compare the resulting files. Both values affect the flow rate, but through different means.

As for the speed override reset on the wipe tower, that was already discussed in #421 - if there are more comments on that from anyone, please, write them to that issue - not here.

tjcolejr, 2018-12-01:
So bottom line, the flow is set based on extrusion multiplier even though it may appear otherwise, and if we are going to monkey with the speed, we need to do it at the outset in speed settings.  Works for me.  Thanks for explaining.

Area5142, 2018-12-10:
To me, it looks like the flow rate adjustment of 95% is a *permanent* setting for MK3, as it is not changed during the print and is used in all prints (layer heights >= 0.075).

If the flow rate adjustment was moved to "Printer Settings" it could be configured for each printer and used in the calculation of the filament feed.
This would also free the *on the fly flow adjustments* in firmware between prints (M221).

When Slic3r calculated the used filament in a print, this is be based on a 100% flow rate (unaware of later firmware adjustments), introducing an error when always lowering firmware flow rate to 95%.
As a printer setting this error could be eliminated.

rtyr, 2020-03-30:
I believe we can close this issue, 95% flow is intentional like Lukas explained.

closed 2020-03-30 by rtyr

wiebel, 2023-01-31:
Sadly the link by @lukasmatena  is dead. The solution however seems like a very hacky workaround. This code made it's way into other printer models as well. It would at least be interesting if the issue addressed by this "solution" is viable only for the MK3.