prusa3d/PrusaSlicer #1447: supports overlap object when 2 STL files loaded

csimmerling, 2018-12-01:
### Version
1.41.2

### Operating system type + version
Windows 7

### Behavior
When I split an STL file into 2 objects and then generate supports, the supports for the first object can overlap and go through the second object. This results in inability to remove supports. I need there to be 2 objects so I can slice for MMU and have each object a different material. It works fine when the stl isn't split (1 object overall, no ability to print different materials though).

is there a better way to print 2 objects in different material, each with supports? there is no way to ensure that supports for each objects do not land on another object.

layer preview when using single object, works fine:
[image]

layer preview with 2 objects,green supports for object 1 are in the middle of object 2:
[image]

_Is this a new feature request?_
no
#### STL/Config (.ZIP) where problem occurs
[fpg.longboth (1).stl.zip](https://github.com/prusa3d/Slic3r/files/2635909/fpg.longboth.1.stl.zip)

FidelCapo, 2018-12-04:
I think this is duplicate of #316.

bubnikv, 2018-12-13:
Duplicate of #316. Closing.

closed 2018-12-13 by bubnikv