prusa3d/PrusaSlicer #1448: Bad surface slicing 1.41.2

Goury, 2018-12-01:
### Version
1.41.2

### Operating system type + version
Ubuntu 16.04 with latest updates and 4.15.0-39-generic kernel, x64

### Behavior
1.41.1 sometimes slices fine
1.41.2 slices badly
Same part from the same file, same settings, same everything, but somehow top layer (sometimes few top layers) are just missing its parts.
But sometimes it just slices fine for no reason.

Steps to reproduce:
* load the file
* lay it on its flat back side
* press "Slice now" button

Expected results:
* Flat top surface

Actual results:
* Scar on parts face
[sliced_1 41 2](https://user-images.githubusercontent.com/1081221/49333521-0ff8df00-f5c9-11e8-82e5-87ab57fe8a00.png)

#### STL/Config (.ZIP) where problem occurs
[report.zip](https://github.com/prusa3d/Slic3r/files/2636373/report.zip)

BerndJM, 2018-12-03:
If you change your first layer height to 0.2 it slices fine in both versions, with your 0.3 it slices bad - also in both versions.
So I asume there is something not 100% flat your top surface - and this confuses the slicer depending on the ratio "sum of the layer heights" vs "real height of the part" ...

BerndJM, 2018-12-04:
I just tried another idea that comes in mind:
1.) rotated your part in NetFabb before loading in Slic3r and voilá it slices fine even with the 0.3 first layer. Can you please try and confirm this with the attached file
[Stand - stand arm 75_rot.stl.zip](https://github.com/prusa3d/Slic3r/files/2642023/Stand.-.stand.arm.75_rot.stl.zip)

2.) rotated the part in SLic3r at 90.01° also slices fine

[image]
from top to bottom: 
- prerotated in NetFabb
- rotated "normal" -90°
- rotated -90.01°

So it seems something goes wrong (may be a rounding error???) while rotating the part in Slic3r.

numanair, 2018-12-13:
To work around this issue I have been adding an offset of +/- 0.01 to the first layer thickness.

bubnikv, 2018-12-13:
This is an evergreen, which should have its own FAQ page on the wiki.

You are asking Slic3r to do a rounding of the object height in the middle of a layer. What is correct? To round below or above? The best is to ease Slic3r's job by giving it objects with a height, which is a multiple of layer height.

I am closing this issue, however I would appreciate, if somebody could create a FAQ on the Slic3r's wiki page.

closed 2018-12-13 by bubnikv

Goury, 2018-12-13:
@bubnikv
Are you kidding?

The correct way is to round flat surface height to be flat after rounding.
Height here is 3.0mm. So either round every 3.0mm vertex to 3.1mm or to 2.9mm.
But not like this, half of the surface is at one height and another half is at another height.

As for which way to choose, there is simple solution: let the user choose.
Like it's done in Cura.

Rounding the same height semi-randomly to different results is clearly not how it should work.

bubnikv, 2018-12-13:
I am not kidding.

> As for which way to choose, there is simple solution: let the user choose.

That is a simple solution for the user, not for the software engineer.

> Like it's done in Cura.

Would you please point me to the parameter?

> Rounding the same height semi-randomly to different results is clearly
not how it should work.

It is not rounding semi-randomly, it is rounding by an exact rule. Just the
vertices are jittered around the slicing plane, only it is not obvious.

DrLex0, 2018-12-13:
Cura does seem to suffer less from this problem but it is not obvious why. I don't find a setting that seems related to this. I need to artificially amplify the Z error around 3mm in my Blender-rotated model with a factor of 1e5 before I get the same effect in Cura.
From a cursory search for its inner workings, Cura seems to rely on integer arithmetic so they probably round coordinates of the imported model to a limited number of decimal places. That would avoid this ‘dithering’ problem in many cases. However, [it will still exhibit this problem](https://community.ultimaker.com/topic/18303-why-is-cura-generating-partial-top-layers/) when the conditions are right.

Goury, 2018-12-13:
However or not, 0.001 mm variations shouldn't result in 0.2 mm difference.
I do understand that this might be not an easy task in some situations (I'm software architect myself), but this is something that have to be done.
I don't ask you to throw away all your weekend plans and start solving my issue.
I ask you to recognize such behaviour as a problem and put it on TODO list.
Let it just be there so people know about existence of this issue and may be someone some day will come with a solution.

cjo20, 2018-12-13:
> However or not, 0.001 mm variations shouldn't result in 0.2 mm difference.
It's taking geometries and slicing them in to discrete 0.2mm layers. This is exactly what has to happen at some point, or you don't end up with layers.

> I do understand that this might be not an easy task in some situations (I'm software architect myself), but this is something that have to be done.
Being a software architect, perhaps you could write a program that take the following list of numbers and tell me which ones are meant to be under 3mm and which aren't? [2.999996, 2.9999999, 2.99994 2.999993]. This is the sort of problem that computers are *bad* at solving.

Sebastianv650, 2018-12-13:
Maybe the Cura option refered here is the "Slicing Tolerance", more details can be found here: https://ultimaker.com/en/resources/52672-experimental

If I understand it right, Slicer would use "Inclusive" mode which means on a model with a height of 0.2mm and a layer height of 0.2mm (to make it simple), Slicer uses the top surface at 0.2mm height for the first slice. As we may get triangles variing around 0.1999999 to 0.2000001 this might result in the problem above.
Curas default is "Middle", so it would use the geometry at 0.1mm height given the example part with 0.2mm height. This way, 0.000xmm deviations are not relevant as long as you slice at usual layer heights.

DrLex0, 2018-12-13:
@Sebastianv650 Slic3r uses Middle. This is why this model ends up being sliced at the problematic Z=3mm height, because first layer height is 0.3 and next layers are 0.2, so the model is cut at 0.15, 0.4, 0.6, …
It doesn't matter though because even with the other strategies, in the end the cut is still done at a fixed height and if there's anything subtly varying around that height, the same problem occurs.