prusa3d/PrusaSlicer #1449: 24 hours added to all estimated print times

CuriousPato, 2018-12-02:
### Version
1.41.2+win64

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### windows 8.1 pro

### Behavior
*all of a sudden att configurations removed from slicer, had to reload all configs from config folder, and many were not the same as they were. estimated printing time now adds 1 day to all prints
* _changed nozzle diameter from 0.3m to 0.8mm
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-12-03:
It is very improbable, that Slic3r would add exactly one day to the print
time estimates for all your prints.
Would you please attach the generate G-code and export your plater into a
3mf file and attach it as well?

bubnikv, 2019-01-29:
No response, closing.

closed 2019-01-29 by bubnikv