prusa3d/PrusaSlicer #1451: Gyroid infill draws some unwanted perimeters

mbuc, 2018-12-03:
### Version
1.41.2

### Operating system type + version
Win10 Pro 64-bit

### Behavior
I am attempting to print objects with just gyroid infill, no shells.  With perimeters set to 0 and top/bottom solid layers set to 0, the gyroid infill still draws lines to connect infill paths where there should be empty space.  An example 20mm cube at 10% gyroid infill:
[image]

I don't know if this would be considered a bug or a new feature request.  I have tested this with many objects at different levels of infill.  The extra lines ruin the visual appeal of the objects I am trying to make.

mbuc, 2018-12-03:
Since I can't search this forked repo, can someone point me in the direction of the code that processes gyroid infill?

bubnikv, 2018-12-04:
Why do you think the connecting lines are wrong? They are there for the
purpose of anchoring the infill to the inner perimeters.

supermerill, 2018-12-04:
@mbuc 

in fillgyroid.cpp, remove the line 183 (the "pts_end.insert")

supermerill, 2018-12-07:
@mbuc I added an option to do what you want on my latest fork release.

all-in-one commit for this "fix" : cb65da2b3af1c8de0e58ff99b7356da28b2f8afc

wavexx, 2019-02-25:
Here's a test picture of a cylinder printed using gyroid without infill connection. 2cm diameter, 3cm height, 9% infill, 0 perimeters, 0 top layers and 1 bottom (for adhesion):

[image]

As mentioned elsemwhere, the gyroid infill is surprisingly sturdy. I couldn't break it with finger pressure alone. Without perimeters though it doesn't result in a perfect print: some walls are started in mid-air, resulting in some stringing. Fortunately, they eventually bridge so any volume seems to be printable if the structure is dense enough.

Adrian-at-CrimsonAuzre, 2019-03-13:
This also happens with other infill types, I would love an option to disable it for doing infill art.

Sembazuru, 2020-04-28:
This issue is still there on PrusaSlicer 2.2.0 (see this forum [thread](url=https://forum.prusaprinters.org/forum/english-forum-general-discussion-announcements-and-releases/exposed-gyroid-infill/)). I'll echo my comment there here:
The caps on the perimeter of the gyroid infill make sense when there are perimeters to help them adhere. I agree, though, the caps shouldn't be there when the infill doesn't abut against perimeters.

josh3io, 2020-10-29:
There are use cases where turning off these perimeters is functionally desirable, not just aesthetic, when other perimeters are set to 0.

bubnikv, 2020-11-20:
The upcoming PrusaSlicer 2.3.0-alpha4 introduces a new config value infill_anchor.
Set it to zero to disable anchoring of sparse infill to innermost perimeter.
Closing.

closed 2020-11-20 by bubnikv