prusa3d/PrusaSlicer #1452: stl failure on SD card but works via Pronteface

reflexxx999, 2018-12-04:
### Version
_Version of Slic3r used goes here_
1.41.2=+win64
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 Pro 10.0.17134 Build 17134
### Behavior
* _Describe the problem_
(1) When trying to use attached stl files in Slic3r Prusa Edition and using more than 2 colors slice now and export G-code causes application to crash. When only using 2 colors stl and g-code is created
(2) 2 color g-code is created and deployed onto SD the SD print fails consistently 10% thru the build seemingly the MMU gets lost and selector believes it's on filament 4 but roller is down on filament 1.
(3) Printing via same g-code is run thru pronteface it completed with challenges since the rolling pin get's misaligned at points and doesn't make contact with filament and feeder just spins. I make correction by pausing and manual rotating the rolling pin and resuming. The pin is securely mounted onto the flat portion of rod so slippage doesn't seem to be the issue.
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
not.. bug report
[test4.zip](https://github.com/prusa3d/Slic3r/files/2642048/test4.zip)
[dragon432_0.15mm_PLA_MK3MMU2.zip](https://github.com/prusa3d/Slic3r/files/2642053/dragon432_0.15mm_PLA_MK3MMU2.zip)

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

reflexxx999, 2018-12-04:
[config.zip](https://github.com/prusa3d/Slic3r/files/2642160/config.zip)

bubnikv, 2018-12-13:
I am afraid this is not a Slic3r issue. If you believe it is a firmware issue, please post it to 
https://github.com/prusa3d/prusa-firmware/issues

If you think it is a mechanical issue, please contact our support.

closed 2018-12-13 by bubnikv