prusa3d/PrusaSlicer #1454: crashes when changing resolutions

theLEDwheel, 2018-12-05:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

FidelCapo, 2018-12-05:
Can you please describe the issue with more details? Which version of slic3r are you using and what is your OS? Can you describe the whole procedure how reproduce a issue? Thank you

bubnikv, 2019-01-29:
No response, closing.

closed 2019-01-29 by bubnikv