prusa3d/PrusaSlicer #1456: Slic3er inconsistently slices model with mirrored faces

dmarra, 2018-12-05:
### Version
1.41.2+win64

### Operating system type + version
Windows 10

### Behavior
When slicing certain thin models, slic3r will create gaps where there should not be gaps, but appears to depend on the axis alignment on the virtual bed. this was noticed while trying to slice lithophanes:

This is on one side:
[image]

This is on the adjacent side:
[image]
 
and a layer showing the inconsitency in slicing:
[image]

As you can see, only one side of the 4 has this problem

This STL was created by generating a lithophane from a grayscale image, then duplicating it in Blender to create an open ended cube. 

Since this is the same lithophane duplicated around a central axis to create a cube, I would expect to see each side have the same (but mirrored) paths. 

#### STL/Config (.ZIP) where problem occurs
[slic3r.configandstl.zip](https://github.com/prusa3d/Slic3r/files/2650699/slic3r.configandstl.zip)

supermerill, 2018-12-06:
The layer view show a hole inside the geometry.
Can you verify it's not in your stl?

dmarra, 2018-12-06:
It's definitely not. Each face of the cube is the same STL. The mesh was rotated around an axis, and exported as a single STL. If there were any gaps, they would be in the corner where the vertices overlap, but do not connect. 

The paths on the upper left corner of the layer view should be a mirror of the paths happening in the lower right.

Sebastianv650, 2018-12-06:
You only attached the single plane, not the full cube. We can't check for the error this way.

supermerill, 2018-12-06:
can you export the .3mf? plater -> export as 3mf

dmarra, 2018-12-06:
Sorry, I provided the incorrect stl file (had tried alot of things).

Attached is the correct STL file to match the screenshots; as well as the requested .3mf file
[correctSTL.zip](https://github.com/prusa3d/Slic3r/files/2653872/correctSTL.zip)

Sebastianv650, 2018-12-06:
I can't reproduce the error slicing it with the exact same profiles given or my own ones. Strange.

dmarra, 2018-12-06:
Oh boy, the most frustrating type of bug amirite?

Are you definitely using the same version of slic3r?

bubnikv, 2019-01-29:
The issue here is, your box consists of 4 intesecting bodies, while they should be merged into a single body using a boolean union operation. This is an exploded view of your model.

[image]

Running your model through the Netfabb fixing service will make the life of Slic3r much easier.
https://tools3d.azurewebsites.net/

bubnikv, 2019-03-11:
I cannot reproduce the issue on the current master. Let's wait if @dmarra could reproduce the issue with the upcoming 1.42.0-beta

closed 2026-09-29 by prusaslicer-bot[bot]