prusa3d/PrusaSlicer #1457: Undefined subroutine set_gui_appctl

stelgenhof, 2018-12-06:
### Version
`git rev-parse HEAD`: b6861a5

### Operating system type + version
Arch Linux: 4.19.2-arch1-1

### Behavior
The application halts immediately with the following message:
`Undefined subroutine &Slic3r::GUI::set_gui_appctl called at slic3r.pl line 112`

Compiled with the latest commits. Also re-compiled with a clean build folder, however still the same issue.

bubnikv, 2018-12-06:
The current master is C++ only. The only perl bindings left are used by the
unit / integration tests of libslic3r.

You should find your Slic3r binary in the build/src directory.

I suppose you will be surprised by the updated UI :-)

bubnikv, 2018-12-06:
I have removed slic3r.pl and updated Build.PL with a note, that the Perl
bindings are now optional to run Slic3r.

stelgenhof, 2018-12-06:
Thanks! I have been compiling it correctly all the time but was still using the perl executable :( I must have missed something, haha.

The binary in the build/src directory works fine! The UI is indeed different. If I may suggest: please normalize all the icons as the design is very inconsistent among them.

Cheers! Sacha

closed 2018-12-06 by stelgenhof

bubnikv, 2018-12-06:
We are certainly working on the icons.

If you have time, please feel free updating the wiki on how to build / run
the Slic3er on Linux.

stelgenhof, 2018-12-06:
@bubnikv I maybe do that :) A while ago I wrote a blog on my website about building it on Arch Linux. Since lots have been changed, I may need to rewrite that (or better: refer to the wiki here...)