prusa3d/PrusaSlicer #1458: Support not generated

scott0122, 2018-12-06:
### Version
1.41.2

### Operating system type + version
Windows 10 x64

### Behavior
Support not generated
My overhang threshold is 20.

My model
[image]
[image]
[image]
[image]
[image]

Model placement
[image]

1.41.0 work fine
[image]

But 1.41.1 and 1.41.2 does not generate support
[image]

#### STL/Config (.ZIP) where problem occurs
[model and config.zip](https://github.com/prusa3d/Slic3r/files/2653296/model.and.config.zip)

supermerill, 2018-12-12:
Same problem for me. It works in 1.41.0 but not in 1.41.1 nor 1.41.2.

[budgetbridge10_small.zip](https://github.com/prusa3d/Slic3r/files/2671906/budgetbridge10_small.zip)

scott0122, 2018-12-13:
@supermerill My temporary solution is to uncheck "Don't support bridges"

supermerill, 2018-12-13:
ohhh, you're right.
If fact, it's the 1.41.0 that has the bug.
Thank you!

bubnikv, 2019-01-29:
So is it a bug?

If "don't support bridges" is enabled, then the bridge is not supported. I suppose that is correct.

supermerill, 2019-01-29:
yes, everything is right now.

bubnikv, 2019-01-29:
Great, closing.

closed 2019-01-29 by bubnikv