prusa3d/PrusaSlicer #1460: Cylinder at 20 degree angle diameter smaller than straight Cylinder

imqqmi, 2018-12-08:
### Version
1.40.1+win64

### Operating system type + version
Windows 7 sp1

### Behavior
When the input geometry is a cylinder standing up, rotated at 20 degrees, the diameter is not the same when printed standing up straight. The diameter is smaller at the 20 degrees angle.
In a test case diameter should be 16mm but is measured 15.4mm after printing. Shrinkage of plastic is about 1.26%. Compensating for shrinkage: 15.4*101.26% = 15.59mm. I used TPC flex 45 filament, but the same is observed using ABS.

gcode supports my finding. I've created two cylinders as described above, copied the outer perimeter and selected a section opposite to each other:

Angle (leaning in the x direction, looking at y coordinates)
G1 X96.355 Y118.769 E8.62924
G1 X96.147 Y103.241 E9.03752
D=15.528 (this is correct, adding external extrusion diameter of 0.45 diameter is 15.978, close enough to 16mm)

Straight
G1 X91.182 Y96.701 E8.49239
G1 X90.988 Y81.329 E8.86180
D=15.372 (15.372 + 0.45 = 15.822. It's 0.178mm narrower than 16mm)

Same test but with Cura, which seems to be correct:
Cura:
Angle
G1 X93.907 Y103.217 E260.32589
G1 X93.928 Y118.784 E260.03230
D = 15.567

Straight:
G1 X92.143 Y96.799 E256.69453
G1 X92.157 Y81.200 E256.39535
D = 15.599

* _Steps needed to reproduce the problem_
Create a cylinder of 16mm diameter in the z direction. Copy the cylinder and rotate 20 degrees so it leans like the tower of pisa. Make an edge with wider diameter for easier reference. Print both objects with the same settings. D_straight != D_at_angle.
Generated gcode
* _Expected Results_
Diameters should at least be the same after slicing. There's a 0.178 discrepancy between straight and at a 20 degree angle (other angles haven't been tested).
* _Actual Results_
Straight cylinder: D=15.528
20 degree angle cylinder: D=15.372

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
[CylinderTestObj.zip](https://github.com/prusa3d/Slic3r/files/2659716/CylinderTestObj.zip)

supermerill, 2018-12-10:
A question

> Cylinder at 20 degree angle diameter smaller than straight Cylinder
and
>Straight : D=15.372 (15.372 + 0.45 = 15.822. It's 0.178mm narrower than 16mm)

which one is too narrow? printed strait or printed with angle=20° ?

imqqmi, 2018-12-10:
I've accidentally swapped straight and at angle, I've updated the OP. Thanks for the head up!

supermerill, 2018-12-10:
Can you do the same for a cube?

imqqmi, 2018-12-19:
I've tried a cube and it came out correctly, so it would seem to be that round objects (irregularly shaped objects?) are affected. Perhaps it's a rounding error? Or a detection of volume where the edge of a square is detected correctly while a cylinder edge volume is much smaller.
I could try a cube again at 45 degrees rotated around the z axis, then measure if the distance is correct angled vs straight.

supermerill, 2018-12-19:
> I could try a cube again at 45 degrees rotated around the z axis, then measure if the distance is correct angled vs straight.

It will help.

imqqmi, 2018-12-19:
I've now tried a cube, rotated at 45 in the z axis, both straight and at 20 degree are correct. 

I've also retried the cylinder test but I can't reproduce the error now. Very strange! Please close this issue. I'll post a new issue if I find a way to reproduce it again. Sorry for the inconvenience.

supermerill, 2018-12-19:
I can't close the issue (i'm not a prusa employee) but you can.

For your problem, you may want to check your geometry, as the cylinder-> 3D model with triangle (.stl) conversion can create some artifacts.

imqqmi, 2018-12-19:
geometry was created in a polygon modeler (lightwave), there's a 1:1 conversion to STL. Loading the STL back into lightwave and measuring it had the correct dimensions. Ok I'll close it for now. Thanks!

closed 2018-12-19 by imqqmi