prusa3d/PrusaSlicer #1461: Concentric infil has gaps that can be filled with straight lines

madsi1m, 2018-12-09:
### Version
1.41.2+win64

### Operating system type + version
Windows Server 2016 x64

### Behavior
When using concentric infill for top+bottom, some times (depending on shape) the concentric infill leaves gaps as shown here:

[image]

It looks like concentric fill needs to check that the layer if full as in the above example clearly shows gaps that should have been filled.

Looking forward to a solution, keep up the good work.

supermerill, 2018-12-10:
Duplicate of https://github.com/prusa3d/Slic3r/issues/577
(and https://github.com/prusa3d/Slic3r/issues/662)
Same as https://github.com/slic3r/Slic3r/issues/4629 ; https://github.com/slic3r/Slic3r/issues/4560

bubnikv, 2019-01-29:
The concentric infill is missing a final gap fill.

supermerill, 2019-01-29:
if you want the code for "final gap fill": supermerill@60442badc2e565c04ed13daba2f9f01c9e7d1766
I can make a pr also.

bubnikv, 2019-01-29:
> if you want the code for "final gap fill": supermerill/Slic3r@60442ba

I appreciate your offer. I am looking forward into merging your extensions,
but we have to roll the current build out of the door first.

madsi1m, 2019-01-29:
Awesome!!

jurriaan, 2019-09-22:
@bubnikv Is this still on your radar? Concentric infill is not really usable in its current state.

bubnikv, 2019-09-23:
It is on the radar, together with the other 938 issues.

madsi1m, 2019-10-01:
@supermerill I think a pull request would push for a fix in an official release a bit faster than not having one.

In the mean time i am using arch curves and faking the center position

stephen304, 2020-05-23:
> if you want the code for "final gap fill": [supermerill/Slic3r@60442ba](https://github.com/supermerill/Slic3r/commit/60442badc2e565c04ed13daba2f9f01c9e7d1766)
> I can make a pr also.
> 
> Note that it uses some code i added with the ironing infill (making the no_sort parameter working)

Is that commit the only one needed to fix? I'm thinking of cherry picking and compiling myself because I need this for printing ear savers for local hospitals as fast as possible.

supermerill, 2020-05-23:
@stephen304 i don't know, fill.cpp ahs changed in the meantime.
You may have to merge some more things from my fork, like the changes for the infill pattern api. Also  I don't really know how well the no_sort is handled in prusaslicer at this time (better than before, for sure).

The basic idea is to transpose the way gapfill is made for perimeters into this. And then add these gap fill extrusions after (and only after) the fill was performed.

tspannagel, 2022-01-01:
This is still an issue. The Gaps produced are really big. 
[image]
In the image you could place 2-3lines in the gaps. (im currently usin pursa slicer 2.4

supermerill, 2022-01-04:
> In the image you could place 2-3lines in the gaps. (im currently usin pursa slicer 2.4

If I copy-paste two lines, they don't fit (but not by much).

tspannagel, 2022-01-04:
superslicer offers a solution for this issue -> "concentric (filled)".  Maybe its possible to merge this feature back?

supermerill, 2022-01-04:
> Maybe its possible to merge this feature back?

It needs to have my new Fill api, to be able to output multiple extrusion types, ordered. Not a very big deal.
I sadly also had to deactivate the greedy reordering and fall back to the 'closest point' one as they don't like when the extrusions are ordered. So for a clean merge, this complicated algorithm has to be enhanced.
If these two are merged then it's just the matter of copying two blocks of code.

closed 2022-06-29 by bubnikv