prusa3d/PrusaSlicer #1464: Crash on cut

supermerill, 2018-12-10:
### Version
Slic3rPE-1.41.2+win32-full-201811221508

### Operating system type + version
windows 10 (64b)

### Behavior
when i try to cut this stl
[catan_main_landscape_ore_newbase_cut.zip](https://github.com/prusa3d/Slic3r/files/2663728/catan_main_landscape_ore_newbase_cut.zip)
 at 0.2mm, slic3r crash to desktop.

It doesn't crash if i try at 2mm.

The given stl was created by a cut and then exported (in slic3rPE).

vojtechkral, 2018-12-17:
Thanks, I'll have a look ...

Panayiotis-git, 2018-12-22:
I have an abandoned MMU print. The printer failed to detect that during the loading of a filament it stuck before the extruder and ruined the print.

I'm trying to fix it by printing the rest of the print and glue the two parts together.

When I cut the model at the height of 36.4mm, Slic3r crashes.
If I try at a lower height (ex. 30mm) it succeeds but then crashes later, when I'm trying to remove the remaining 6.4mm.

I attach the 3mf plate export.

[Gustav the Turtle_0.2mm_PLA_MK3MMU2.zip](https://github.com/prusa3d/Slic3r/files/2705188/Gustav.the.Turtle_0.2mm_PLA_MK3MMU2.zip)

bubnikv, 2019-01-29:
Most likely the cut function crashes due to the triangulation library not being numerically robust to degenerate input.

bubnikv, 2019-02-05:
The crash now is fixed on master by swapping the triangulation library with a proven glu tesselator (written in 1994 in plain C by SGI).

While the triangulator does not crash, if the input mesh is broken, the cut is not always triangulated. Namely, inverted parts are not triangulated.

Closing.

closed 2019-02-05 by bubnikv