prusa3d/PrusaSlicer #1465: BUG MMU code adds excessive changes to the purge block

johnbiggs, 2018-12-10:
### Version
1.41.2 Release

### Operating system type + version
OSX

### Behavior
The MMU logic adds extra color changes for multiple STLs which share the same extruder assignment.
* _Steps needed to reproduce the problem_
Load an STL with lots of parts. Assign parts through the extruder range. If the STL has enough parts the purge block no longer fits on the bed.
  
*_Expected results
Purge block is the same max size with a 5 color print based on the purge values. Treat multiple STLs assigned to the same extruder as a single STL.
* _Actual Results_
 Purge block continues to grow with more parts although the number of colors haven't increased. I believe the Slicer is intending excessive transitions.

_Is this a new feature request?_
No

BTW: the format of this is strange. I wasn't sure if I should keep the comments or delete the comments. Contributing Guidelines weren't clear.

bubnikv, 2018-12-13:
Please provide a 3MF (go to menu Plater -> Export as 3mf), and attach some screen shots documenting your issue. Thanks.

bubnikv, 2019-01-29:
No response, closing.

closed 2019-01-29 by bubnikv