prusa3d/PrusaSlicer #1466: [Feature request] Per filament "extra loading distance"

Flav1u, 2018-12-10:
Using the MMU2 with non-flex and flex filaments would be easier with the possibility to set "extra loading distance" per filament (or just for flexibles).

supermerill, 2018-12-13:
maybe you can add that line in the (flexible) filament settings -> custom g-code -> start g-code
something like ''G1 E9.0  F1000.0"

bubnikv, 2018-12-13:
@akukan was planning to implement the extra distance into the MMU2 unit's firmware based on the filament type.
@akukan what is the current state of the feature?

rtyr, 2021-01-22:
This is firmware issue. PrusaSlicer already sends info about some particular filament types (FLEX/PVA), but the firmware currently ignores this information.

Closing.

closed 2021-01-22 by rtyr