prusa3d/PrusaSlicer #1467: Internal parts not being sliced

jsilva-iii, 2018-12-11:
### Version
1.41.2+win64

### Operating system type + version
Win 10

### Behavior

Part not being sliced correctly internal parts are missing

#### STL/Config (.ZIP) where problem occurs
[image]
[image]
[Config & STL.zip](https://github.com/prusa3d/Slic3r/files/2665481/Config.STL.zip)

Sebastianv650, 2018-12-11:
The stl in your attachment is only a 2kB "big" plate.

jsilva-iii, 2018-12-11:
Hi

Sorry, I'll attach the correct one when I get home, or I can email to this.

Thanks
Jorge

jsilva-iii, 2018-12-13:
> I think the missing part of the object is very thin, because when I turn
> on detect thin walls, everything seems to be generated correctly. Can you
> confirm this? Thank you

Hi

I can confirm that if I turn on thin walls it slices correctly, I’ll check
the design as it was supposed to be 0.9mm thick. Thanks for the help.

Thanks
Jorge