prusa3d/PrusaSlicer #1469: Bridging logic

ijustwannafly, 2018-12-11:
I have been using slic3r prusa edition as my primary slicer.

I have noticed on bridging moves the hotend goes straight into the bridge path from it's starting/entry point.

Even with all the tuning I always manage to get a bit of droop at the start of the bridge just before the filament catches up to the speed and extrusion changes set for the bridge.

Is it possible to add a logic that drags the hotend along the printed surface (infill area of previous area) while allowing the hotend a small amount of time to get the speed and extrusion settings for bridging set before actually going into the bridge move?

Sebastianv650, 2018-12-11:
This will not solve the problem, maybe even make it worse. The issue you see as a droop is a sudden change in the nozzle pressure. When you print normal above the last layer, we have a certain amount of backpressure + the pressure we would have during extrusion in free air.
When the nozzle moves away from the perimeter into free air, this "overpressure" unloads into a sudden "overextrusion" which droops down.

If you switch to bridging parameters before the bridging actualy starts, the pressure + layer backpressure will be even higher as the bridging flow is handled as a full string of nozzle diameter.

ijustwannafly, 2018-12-11:
So from your comment should there be another setting like "pre-bridge" retraction to help alleviate some of that pressure just before getting to the bridge entry point?

Sebastianv650, 2018-12-12:
A retraction would interrupt the movement, which is also bad. If we adjust the flow rate to some "equivalent pressure rate" this will lead to over or underextrusion before the bridge. I don't think there is a good solution available.

ijustwannafly, 2018-12-12:
So could a logic be done to where instead of going straight into the bridge entry point the nozzle would lift z by 0.1mm, retract, then descend onto the bridge entry point while de-retracting all while moving?

Almost like a movement going over a hill?  As you finish the last point of the previous layer make a move to start moving and lifting z while retracting and then while still moving descend onto the the bridge entry point while de-retracting?

Just thinking out loud.

Sebastianv650, 2018-12-12:
There is just one method to see what will work and what not: Create a bridge test gcode and modify the bridging line gcode by hand to simulate what slic3r should do later in an automatic way. I don't think they will code new features without knowing if it will even work as expected.

closed 2026-09-29 by prusaslicer-bot[bot]