prusa3d/PrusaSlicer #1471: Modifier mesh reload broken: Slic3r "Reload from disk" - UI function overhaul #4388

fusentasticus, 2018-12-12:
### Version
1.41.2
### Operating system type + version
all
### Behavior
Modifier meshes are disregarded upon reload of part
### Related Observations
* There is no way of starting Slic3r with a command line argument to specify modifiers
* The .amf format is a possible workaround but seems to require merging of vertices in all meshes and triangle renumbering
* The problem is mentioned in this thread: https://github.com/prusa3d/Slic3r/issues/552, which had another focus
* The problem has been solved in the original branch https://github.com/slic3r/Slic3r/pull/4388

I'm hoping you can address this issue! The support modifiers are very nice but it is very impractical to design with them currently.

bubnikv, 2018-12-13:
This is something for @vojtechkral to look into.

bubnikv, 2019-12-18:
Reload from disk has been reworked for PrusaSlicer 2.2. Closing.

closed 2019-12-18 by bubnikv