prusa3d/PrusaSlicer #1472: Slic3r drains all memory available and crashes

mkbel, 2018-12-13:
### Version
_Version of Slic3r used goes here_
Slic3rPE-1.40.0-alpha1+14-fwupdater-linux64-full-g9223ba7-201805161352.AppImage
Slic3rPE-1.40.0-alpha1.linux64-full-201805141245.AppImage
Slic3rPE-1.40.0.linux64-full-201806121015.AppImage
Slic3rPE-1.40.1-beta.linux64-full-201806220828.AppImage
Slic3rPE-1.41.0-alpha2+linux64-full-201807211541.AppImage
Slic3rPE-1.41.0-alpha3+linux64-full-201807272049.AppImage
Slic3rPE-1.41.0-beta+linux64-full-201808061625.AppImage
Slic3rPE-1.41.1-rc+linux64-full-201810051337.AppImage
Slic3rPE-1.41.2+linux64-full-201811221508.AppImage

### Operating system type + version
Debian testing 64bit

### Behavior
* _Describe the problem_
Slic3r drains all memory available and crashes
(OS becomes unresponsive when not limited on memory)
(itself with:
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
if memory of its process is limited to 6GB)

* _Steps needed to reproduce the problem_
open stl  [vymenik.stl.zip](https://github.com/prusa3d/Slic3r/files/2674461/vymenik.stl.zip)
select print settings 0.20 mm FAST MK3
set Perimeters to 0
set infill to 100%
click Slice now

_Is this a new feature request?_
NO

#### STL/Config (.ZIP) where problem occurs

[config_vymenik.ini.zip](https://github.com/prusa3d/Slic3r/files/2674465/config_vymenik.ini.zip)

bubnikv, 2018-12-13:
On my 3 year old 4 core I7 Windows 10 laptop with 16GB of RAM, and testing the current 1.42 alpha, I get the slicing finished very quickly, with the G-code export taking the longest time.

[2018-12-13 08:50:15.763366] [0x00000228] [info]    Exporting G-code...
[2018-12-13 08:50:50.759241] [0x00000228] [debug]   Preparing G-code preview data
[2018-12-13 08:50:51.881598] [0x00000228] [info]    Exporting G-code finished

You can allow this kind of debugging output by setting the env variable SLIC3R_LOGLEVEL to 1 to 9.

After the slicing, Windows system reports the following statistics on memory usage:
[image]

What strikes me is the virtual size of the process, which is 8GB. Sometimes the graphics driver's memory is being mapped into the virtual memory space of the process, sometimes it is the stack allocated per each running thread, which increases the virtual memory allocation excessively.

A lot of RAM is consumed by the OpenGL vertex buffers for the 3D G-code path preview, but it is not that bad for your model:

[image]

The virtual memory size shown is 8GB, but from that 6GB are "reserved" virtual memory used likely by some Windows security mechanism. Total committed memory is around 1.5GB on my system after slicing with the G-code preview on.

vojtechkral, 2018-12-13:
I was able to reproduce the problem on my linux box.

mkbel, 2018-12-13:
The problem can be also reproduced on:
Slic3rPE-1.42.0-alpha+952-gui-transforms-linux64-full-g57ef08f-201812131401.AppImage
But behaves little different. It doesn't utilize all 8 logical CPU cores I have, but just one core. It takes a lot longer to crash.
Doesn't crash immediately, after slicing std::bad_alloc is shown on bottom bar.
It crashes after clicking on preview.
writing: "Unhandled unknown exception; terminating the application."
on terminal.

mkbel, 2018-12-13:
It is possible to slice the same object successfully on the same machine (8GB RAM) on Windows 10 with Slic3r 1.41.2+win64. But one can not use preview. If just object is added and export Gcode is clicked, Gcode is exported fine.
If object is added, Slice now is clicked, preview is activated, then Export G-code is clicked, gcode is generated only partially and pop-up window containing Bad Allocation is shown.
[image]

bubnikv, 2018-12-17:
77d37f108c3ba68afaebda9f54933f2b4185aaab implements an iterative (non-recursive) Douglas-Peucker-Ramer algorithm based on the work by @fuchstraumer (pull request upstream)

Would you please retest with Slic3rPE-1.42.0-alpha+949-linux64-full-g9fbbbca-201812170810.AppImage
It finished successfully on an Ubuntu 16 virtual machine with 8GB RAM and 6 virtual CPUs assigned.

mkbel, 2018-12-17:
Problem is not fixed in Slic3rPE-1.42.0-alpha+949-linux64-full-g9fbbbca-201812170810.AppImage. Tried with 8GB soft memory limit enforced by ulimit.

mkbel, 2018-12-17:
Problem is nor fixed in Slic3rPE-1.42.0-alpha+967-linux64-full-gb613334-201812171505.AppImage.

mkbel, 2018-12-17:
My bad, configuration initially supplied wasn't able to reproduce problem. Here is configuration able to cause the problem. [config_vymenik_corr.ini.zip](https://github.com/prusa3d/Slic3r/files/2687411/config_vymenik_corr.ini.zip)

bubnikv, 2018-12-18:
I have added memory tracking (getting the process statistics from the OS works on WIN32 only as of now, and it is enabled with the SLIC3R_LOGLEVEL env variable).

After slicing, before G-code export:
Exporting G-code... WorkingSet: 712MB PrivateBytes: 691MB Pagefile(peak): 691MB(691MB)

After the last slice was exported into a temp G-code:
[2018-12-18 16:45:14.776329] [0x00006c5c] [trace]   Exported layer 649 print_z 130, time estimator memory: 3,229MB, analyzer memory: 3,709MB

G-code preview data also consumes some memory.
[2018-12-18 16:46:06.745661] [0x00006c5c] [debug]   Preparing G-code preview data

The final memory consumed:
[2018-12-18 16:47:43.787880] [0x00006c5c] [info]    Exporting G-code finished WorkingSet: 9,378MB PrivateBytes: 15,492MB Pagefile(peak): 15,492MB(16,573MB)

The time estimator memory (3,229MB in your case) was allocated for both the normal and silent mode, so you may save a bit of RAM, if you disable silent mode time estimates. 

We will not improve the RAM consumption of the G-code export in the near future, but we shall do that once time allows.

bubnikv, 2018-12-18:
@fuchstraumer I am aware of the shrink_to_fit() function, but I am not sure how much it will help in case of short polygons. I suppose shrink_to_fit() will do a new memory allocation, which itself is costly, so I would rather avoid that and leave it to the caller to call shrink_to_fit() if necessary.

bubnikv, 2018-12-18:
I have lowered the peak memory consumption with 8bc04e640aae1b45de9f987b826e2e04aa0c601f
by reshuffling the code to release the time estimator memory before generating the G-code preview data, the gain is roughly 1/3.

mkbel, 2018-12-19:
It was proved by @zavorka that memory consumption is not higher on Linux than on Windows, object can be sliced on Linux machine with 32GB RAM.

mkbel, 2018-12-19:
I fixed problem with my swap device (moved it to different partition). Now my system doesn't hangs when swapping. But slic3rPE is crashing when virtual memory starts to be used:

marek@debian:~/bin/Slic3r$ ./Slic3rPE-1.41.2+linux64-full-201811221508.AppImage Creating main frame...
(slic3r.pl:3201): Gtk-CRITICAL **: 17:52:48.223: IA__gtk_color_button_set_color: assertion 'color != NULL' failed
corrupted double-linked list
Aborted

marek@debian:~/bin/Slic3r$ ./Slic3rPE-1.42.0-alpha+952-gui-transforms-linux64-full-g57ef08f-201812131401.AppImage 
[2018-12-19 17:46:20.178423] [0x00007f91d1989400] [error]   Error in a Vendor Config Bundle "/home/marek/.Slic3rPE-alpha/vendor/PrusaResearch.ini": The printer preset "sla_material:Material 1" contains the following incorrect keys: layer_height, which were removed
[2018-12-19 17:46:20.178458] [0x00007f91d1989400] [error]   Error in a Vendor Config Bundle "/home/marek/.Slic3rPE-alpha/vendor/PrusaResearch.ini": The printer preset "sla_material:Material 2" contains the following incorrect keys: layer_height, which were removed
Creating main frame...
Multisample enabled
free(): invalid pointer
Aborted

mkbel, 2018-12-20:
Slic3rPE-1.42.0-alpha+999-linux64-full-g8bc04e6-201812181648.AppImage FAILED free(): double free detected in tcache 2 Aborted
Slic3rPE-1.42.0-alpha+999-linux64-full-g8bc04e6-201812181648.AppImage FAILED free(): double free detected in tcache 2 Aborted
Slic3rPE-1.42.0-alpha+967-linux64-full-gb613334-201812171505.AppImage FAILED free(): double free detected in tcache 2 Aborted
Slic3rPE-1.42.0-alpha+952-gui-transforms-linux64-full-g57ef08f-201812131401.AppImage PASSED
Slic3rPE-1.42.0-alpha+952-gui-transforms-linux64-full-g57ef08f-201812131401.AppImage FAILED double free or corruption (out) Aborted
Slic3rPE-1.42.0-alpha+949-linux64-full-g9fbbbca-201812170810.AppImage FAILED free(): double free detected in tcache 2 Aborted
Slic3rPE-1.42.0-alpha+327-dev-native-linux64-full-gf2b0904-201810300829.AppImage FAILED free(): double free detected in tcache 2 Aborted
Slic3rPE-1.42.0-alpha+325-vb-precompiled-headers-linux64-full-g6539c54-201810291605.AppImage FAILED free(): double free detected in tcache 2 Aborted
Slic3rPE-1.42.0-alpha+320-dev-native-linux64-full-g35d984b-201810261632.AppImage SKIPPED
Slic3rPE-1.42.0-alpha+318-dev-native-linux64-full-g135ee60-201810261440.AppImage SKIPPED
Slic3rPE-1.42.0-alpha+207-dev-native-linux64-full-g30a130e-201810250838.AppImage SKIPPED
Slic3rPE-1.42.0-alpha+103-dev-native-linux64-full-gd843f1a-201810120938.AppImage SKIPPED
Slic3rPE-1.42.0-alpha+20-dev-native-linux64-full-ga9f5295-201809241333.AppImage SKIPPED
Slic3rPE-1.42.0-alpha+16-dev-native-linux64-full-g60a192e-201809211956.AppImage SKIPPED
Slic3rPE-1.42.0-alpha+14-dev-native-linux64-full-g49c3d59-201809211813.AppImage SKIPPED
Slic3rPE-1.42.0-alpha+9-dev-native-linux64-full-g8841b7a-201809211401.AppImage SKIPPED
Slic3rPE-1.41.2+linux64-full-201811221508.AppImage PASSED
Slic3rPE-1.41.2+linux64-full-201811221508.AppImage PASSED
Slic3rPE-1.41.2+linux64-full-201811221508.AppImage PASSED

mkbel, 2018-12-21:
If I use actual master 56f473c2dca318f73a7913f04b74e3f55c30ba88 I build myself. It PASSED.

bubnikv, 2019-01-04:
Please test with Slic3rPE-1.42.0-alpha1+38-linux64-full-gb82f331-201901031756.AppImage
There was a bug in the contour decimation, which may have lead to an excessive memory allocation.

mkbel, 2019-01-10:
Retested with Slic3rPE-1.42.0-alpha2+linux64-full-201901041550.AppImage.
gcode is SUCCESSfully generated.

But I can't use preview, when I click preview button, slic3r is killed due to system memory starvation.

~~~
DRI_PRIME=1 ionice -c 3 nice -n 19 ./Slic3rPE-1.42.0-alpha2+linux64-full-201901041550.AppImage 
Creating main frame...
Multisample enabled
[2019-01-10 19:17:53.831928] [0x00007f2b057fa700] [error]   The downloaded index /home/marek/.Slic3rPE-alpha/cache/PrusaResearch.idx-update for vendor Prusa Research is older than the active one. Ignoring the downloaded index.
Killed
~~~

~~~
dmesg -T| grep -E -i 'killed process'
[Thu Jan 10 19:25:26 2019] Killed process 9189 (slic3r) total-vm:17258344kB, anon-rss:744208kB, file-rss:0kB, shmem-rss:0kB
~~~

I have 8GB RAM and 8GB swap.

I think this ticket can be closed, as main problems like:

corrupted double-linked list Aborted
free(): invalid pointer Aborted
free(): double free detected in tcache 2 Aborted

were solved.

bubnikv, 2019-01-10:
Cool, thanks.

closed 2019-01-10 by bubnikv