prusa3d/PrusaSlicer #1474: [Feature Request] Separate Top/Bottom Fill Pattern Settings

photog0411, 2018-12-13:
I would like the option in Slic3r to use a different top or bottom fill pattern, not to have a single option to use the same for both. 

This is a long shot and you guys probably will reject this request and I hope you are OK with me making so many requests. :) But this will be especially useful with Prusa Printers, and I'll tell you why.

I've found "hilbert curve" to give the best looking top/bottom fill pattern, but it's slower that rectilinear, and it's also prone to causing the first layer to not stick to the bed. When using the Prusa Poweder Coated sheet however, rectilinear for the first bottom layer is the best because the powder coated texture hides the layer lines that you would normally get with rectilinear fill pattern.

So for the best results possible, I think being able to use Hilbert Curve for the top fill pattern and rectilinear for the bottom layer (first layer) fill pattern will give the best results, especially with powder coated prusa bed!

What do you think, possible?

supermerill, 2018-12-14:
duplicate of https://github.com/prusa3d/Slic3r/issues/475

codiac2600, 2018-12-14:
Yes please!

bubnikv, 2019-02-22:
Implemented with ab30370fb4bfa92d8ae77419686e2b6cde923fcf
Thanks @supermerill for the initial implementation.
Closing.

closed 2019-02-22 by bubnikv