prusa3d/PrusaSlicer #1475: Load Config bundle Failed

OmensArea51, 2018-12-14:
### Version
_Version of Slic3r used goes here_
1.41.2+win64

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10

### Behavior
* _Describe the problem_
Exported a configuration bundle file.  Worked
Load the same configuration bundle file.  Failed
* _Steps needed to reproduce the problem_
As above.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
Failer to load configuration bundle.
  * _Screenshots from __*Slic3r*__ preview are preferred_

[SHARON_config_Bronze_005.zip](https://github.com/prusa3d/Slic3r/files/2679743/SHARON_config_Bronze_005.zip)
[Failer.zip](https://github.com/prusa3d/Slic3r/files/2679744/Failer.zip)

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-12-18:
@FidelCapo analyzed the config bundle by @OmensArea51. Your config bundle contains a duplicate profile "print:0.35mm FAST 0.6 nozzle". Indeed, Slic3r complains about a duplicate section, that is [print:0.35mm FAST 0.6 nozzle] (a section of an ini file is a block separated with a text between []).

How did you get there? Is it possible, that your Slic3rPE home directory contains that particular profile "print:0.35mm FAST 0.6 nozzle" stored as your user profile? That should not happen.

Patrik-svobodik, 2018-12-22:
@bubnikv could you please analyze my config? 
[Slic3r_config_bundle.ini.zip](https://github.com/prusa3d/Slic3r/files/2705225/Slic3r_config_bundle.ini.zip)

bubnikv, 2019-01-29:
The config bundle was corrupted, and we don't know how it was created.
We hope that the current Slic3r-1.42.0-alpha4 behaves better: It does not export system profiles, and it ignores duplicate profiles during loading.

Closing.

closed 2019-01-29 by bubnikv