prusa3d/PrusaSlicer #1476: BUG: Bridging and gap fill are parsed incorrectly when infill is set to 0

codiac2600, 2018-12-14:
_1.41.2+_
Max OSX

When using 0% infill bridging doesn't connect but also the gap fill turns off. These two should stay as they are.

 
[image]
[image]
[image]
[image]

supermerill, 2018-12-17:
line 296 in perimeterGenerator.cpp : delete the ' && this->config->fill_density.value > 0' part

currently, the gapfill is deactivated if 
 * gap fill speed is 0

OR
 * infill % is 0  

I don't see the need of the second option.

codiac2600, 2018-12-17:
Gapfill should have an on off check box. It's not intuitive to turn Gapfill off via setting it's speed to zero.

supermerill, 2018-12-17:
Well, the need to disable gapfill (mostly for high-flow extruders that don't like travels) is very small. I don't see the need of a mostly unused setting. If the tooltip of the gapfill speed option is changed, i think (imo) it's enough.

bubnikv, 2019-09-09:
> I don't see the need of the second option.

Maybe because the vase mode

bubnikv, 2019-09-09:
It is a bit more complex than that. In the perimeter generator, you want to enable gap fill in between perimeters, but you want to disable gap fill between the inner most perimeter and the infill area.

For spiral vase, the number of perimeters shall be set to 1 and infill percentage shall be set to zero, so this logic should be sound.

bubnikv, 2019-09-10:
Fixed with 246dc64c99250449b709618bf0816af7d64831e6

The gap fill was disabled for zero infill.
Now the gap fill is enabled in between the perimeters, but disabled between
the inner-most perimeter and infill in case the infill is set to zero. This logic shall work for vase mode as well.

Also in case there are multiple infill regions inside a perimeter, 
the mutliple infills are considered as non-zero if at least one infill
is non-zero, therefore the gap fill will be added inside the inner-most
perimeter.

The fix will be part of PrusaSlicer 2.2.
Closing.

closed 2019-09-10 by bubnikv