prusa3d/PrusaSlicer #1478: Problema

DiegoAlex67, 2018-12-15:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Buongiorno,  uso W10 PRO. In SLIC 3 R V. 1.41.2. Caricato l'oggetto risulta decentrato nonostante le quote del piano siano corrette. La cosa strana che che due oggetti che hanno gli stessi ingombri uno è centrato, l'altro no. Inoltre la funzione Editing degli strati e anteprima non funzionano.

[FILE.zip](https://github.com/prusa3d/Slic3r/files/2682593/FILE.zip)

bubnikv, 2018-12-17:
@enricoturri1966 Would you please help with the investigation? Thanks.

enricoturri1966, 2018-12-17:
Buongiorno DiegoAlex67
Ho provato ad importare i suoi due files in Slic3r (e in altri software).
I due modelli "Scatola multimetro.stl" e "Coperchio multimetro.stl" in realta' non hanno lo stesso ingombro.
"Scatola multimetro.stl" contiene un rettangolo alla base piu' largo delle dimesioni del coperchio e questo
e' il motivo per cui i due modelli non risultano allineati.
E' molto probabile che rimuovendo quel rettangolo dal modello (oppure scalandolo in modo che combaci alle dimensioni della scatola) sara' in grado di importare i due modelli in Slic3r con le desiderate posizioni relative.
Vorrei inoltre suggerire di rimuovere da entrambi i modelli le scritte (a meno che lei non intenda davvero stamparle).

[image]

[image]

closed 2018-12-18 by enricoturri1966