prusa3d/PrusaSlicer #1479: >1.5mm of supports built without adding to purge block or changing filaments to print model (i3 MK3 MMU2.0 using support enforcers)

ea69, 2018-12-15:
### Version
1.41.2

### Operating system type + version
Win10 64bit

### Behavior
two instances of a multi file model are plated. each instance has unique extruder assignments to create one white and one black minion chess pawn. each instance uses three support enforcers. the behavior described below happens on both instances.

the bottom 3.25mm of each model (base.stl) prints correctly with a single filament (F3). An additional > 1.5mm of support then prints using that filament. The purge tower is not built up along with the support material as you would normally expect.

At the Z offsets in which the support material was built, the shoe models should have also been printed, but were not. When a filament change does occur to print the shoes (F1 for once instance and F2 for the other), the filament does not adhere to the purge block during the change nor to the model.

I manually stop the print at this point.

The attached zip file contains:
     all model .stl files
     exported configuration
     generated gcode
     picture of the printbed at the time the print was stopped

[pawn.zip](https://github.com/prusa3d/Slic3r/files/2683135/pawn.zip)

lukasmatena, 2018-12-16:
Thanks for the report. I created a 3MF for everyone to reproduce the issue easily: [pawn_3mf.zip](https://github.com/prusa3d/Slic3r/files/2683381/pawn_3mf.zip)

The two problems are connected, as you probably guessed.
First problem: The shoes are not sliced for some reason, unknown to me at this point. This results in empty layers in the print.

Second problem: Slic3r does not expect these empty layers and the wipe tower is not generated correctly. We already have a fix for this (#1266), it's just not in the master branch yet because it needs more testing. Anyway, this is a consequence of the first problem in your case. Solving it will not help you.

It is therefore necessary to find out why the shoes are not sliced. It clearly has something to do with the support enforcers - removing the support enforcers and using autogenerated supports  makes the issue disappear. Just removing the shoes propagates the issue to the trousers (they do not appear in the ouput). Maybe @bubnikv could see where the problem is.

lukasmatena, 2019-12-17:
This seems to have been fixed long ago between Slic3rPE 1.41 and PrusaSlicer 2.0.0. Closing.

closed 2019-12-17 by lukasmatena