prusa3d/PrusaSlicer #1481: Start GCode strange

ravex1337, 2018-12-16:
### Version
1.41.2+win64

### Operating system type + version
Windows 10, 64Bit

### Behavior
i put my start gcode from cura in and started printing.
i was expecting slic3r to behave like so:
- go up
- extrude
- start printing (move from the current position to the first printing spot, so it moves diagonally down to the printbed)

but instead it does the following:
- go up
- extrude
- go down to 0.2mm
- slam my printbed holding clamp
- do weird stuff

_Is this a new feature request?_
combine the first layer height with the first moving command, see my start code:

G28 -> my start code starts here
G21
G90
M82
M107
G1 Z15.0 F9000
G92 E0
G1 F200 E15
G92 E0
G1 F9000
M420 S1
M900 K75
M117 Druckvorgang... -> and ends here
; Filament gcode 
M109 S197 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 E-4.35000 F6000.00000
G92 E0
G1 Z0.200 F10800.000 -> here the printer moves down first
G1 X77.828 Y67.320 Z0.200 -> and here goes the first printing instruction

i want it to combine these two lines so it moves diagonally down from where it extruded the priming amount, so it does not slam my print bed again..
may i change this in slic3r or do i have to edit my gcode every time like this:
G1 X77.828 Y67.320 Z0.200
?

jornbyte, 2018-12-16:
my start G code: 
Drive all axes home
lift z axes
heat nossel
extruded filament
start printing

G28; home all axes
G1 Z20 F200; lift nozzle
M109 S [first_layer_temperature]
G92 E0
G1 E35 F400

StephenRC, 2018-12-17:
For the start gcode on my duet powered printer, I just use:

T0 ; make sure the hotend is selected
M98 P/macros/PLA offset  ; set z probe offset for pla
M190 S[first_layer_bed_temperature] ; heat the bed and wait
M104 S[first_layer_temperature] ; start heating the hotend
G28 ; home all axes
G32 ; bed calibration - RRF
M109 S[first_layer_temperature] ; finish heating the hotend

I also have set in slic3r to one loop/20mm skirt, for the initial
hotend prime.

On Sun, 16 Dec 2018 07:08:06 -0800, ravex1337
<notifications@github.com> had a flock of green cheek conures squawk
out:

> ### Version
> 1.41.2+win64
>
> ### Operating system type + version
> Windows 10, 64Bit
>
> ### Behavior
> i put my start gcode from cura in and started printing.
> i was expecting slic3r to behave like so:
> - go [...]

-- 

Stephen

. . . . File not found. Should I fake it? Y/N

---
This email has been checked for viruses by AVG.
https://www.avg.com

bubnikv, 2019-08-01:
Likely not a slicer issue. No discussion over the last 6 months, closing.

closed 2019-08-01 by bubnikv