prusa3d/PrusaSlicer #1482: Strange behavior on bridge detection

scott0122, 2018-12-18:
### Version
Slic3r PE 1.41.2+win64

### Operating system type + version
Windows 10

### Behavior
My model look like this
[image]

I choose to generate support with contact Z distance 0. Please see the part indicated by the arrow below.
[image]

Then I canceled the support. The support disappeared, but the bridge's fill pattern did not change.[image]

Then I modify the fill density and the fill mode of the bridge is back.
[image]

#### STL/Config (.ZIP) where problem occurs
[bridge_test.zip](https://github.com/prusa3d/Slic3r/files/2689836/bridge_test.zip)

bubnikv, 2019-01-29:
It seems like an update issue of the slicing back end. Would you please retest with the current Slic3r 1.42.0-alpha? We did a lot of changes at the back end to enable reliable background processing.

bubnikv, 2019-04-13:
no response, closing

scott0122, 2019-05-24:
Still not working on 2.0 release

Test by the following steps:

support_material_contact_distance = 0, support_material = 1
[image]

support_material_contact_distance = 0.2
[image]

support_material_contact_distance = 0
[image]

support_material = 0
[image]

bubnikv, 2019-05-24:
support_material_contact_distance means soluble supports.

bubnikv, 2019-09-09:
I can confirm that there is an update problem: The back-end gets invalidated if the support_material_contact_distance value changes, but the back-end does not get invalidated to enable bridging if the supports get disabled (by setting support_material=0) while having support_material_contact_distance=0 (meaning soluble supports).

bubnikv, 2019-09-09:
Fixed with 5c72eecf26a63868ea04751a956f6088fc58f0e7
The fix will be part of 2.2 release.
Closing.

closed 2019-09-09 by bubnikv