prusa3d/PrusaSlicer #1487: Slicing multicolor objects with sharp corners or fine details produces little holes in the result

Rick-T, 2018-12-20:
### Version
1.41.2

### Operating system type + version
Manjaro Linux (latest)

### Behavior

Slicing multicolor objects with sharp corners or fine details produces little holes in the result.

To reproduce, slice this: https://www.thingiverse.com/thing:3299821

**Expected result:** The object is sliced with all the colors "fitting together".

**Actual result:** On small details and sharp corners there are little holes in the sliced object. These appear only at the color boundaries. Enabling thin-wall detection does not help. Instead it fills some holes, but creates others in different locations.

**Screenshots:**

*3D-View*
[image]

*Preview*
[image]

*Preview with thin-walls*
[image]

#### STL/Config (.ZIP) where problem occurs

[config.zip](https://github.com/prusa3d/Slic3r/files/2698560/config.zip)

Rick-T, 2019-01-30:
With "Detect thin walls" it's different, but not really better. Some holes will be filled by "Detect thin walls" but there will also be new holes that weren't there before.

OP has one screenshot with "Detect thin walls" and one without, so you can compare them directly.

scooper4711, 2020-06-06:
+1 
It’s annoying and even clips out some details in favor of a hole. :(

closed 2026-09-29 by prusaslicer-bot[bot]