prusa3d/PrusaSlicer #1493: Feature request: Configure wipe to infill/object settings on individual parts

jettoblack, 2018-12-23:
There are cases where it would be nice to be able to configure wipe to infill and wipe to object on individual object parts instead of the whole object:

* Wipe to object on parts that are for support only (e.g. Meshmixer tree supports) or internal geometry which isn't visible on the final object would avoid a lot of unnecessary filament changes when the selected filament wouldn't be needed on that layer

* Wipe to infill could be disabled on parts with a light colored or translucent perimeter, so that the infill wipe doesn't show through; and then enabled on parts with a dark colored perimeter since lighter infill won't show through

lukasmatena, 2018-12-23:
When I implemented the wipe into object/infill feature, I didn't allow setting wipe into OBJECT for individual parts for some technical reasons. I don't recall all details this quickly, but I think it would have added a lot of complexity to the code, so we decided not to do it. We may come back to it one day.

As for the tree-supports situation, it should be possible to open the model twice (so the two models overlap), remove the support subpart from the first and everything but the support part from the second and then setting it the way you want. It is a nasty workaround, but seems to be working. You can also use a modifier mesh to override the extruder on some part of the supports.

Wiping into INFILL can actually be set for individual parts, it's just quite complicated to do in the current UI. You have to open the object settings dialog, click on the subpart, click the "plus" icon to override a setting and choose Extruders->Wipe into this object's infill. This allows you to set the value to whatever you want to. The next release 1.42 will make the job easier - an alpha is to be released shortly.

jettoblack, 2018-12-25:
Thanks @lukasmatena this is very helpful info.

Canyonazo, 2019-05-30:
Does wipe into infill/object work in PrusaSlicer? I am not able to enable it :( @lukasmatena

lukasmatena, 2019-05-30:
As far as we know, it does. If you think otherwise, you have to be more specific than this.

bubnikv, 2019-06-03:
You have to open a pop-up menu over the object in the object list and
select the "Extruders" item.

closed 2026-09-29 by prusaslicer-bot[bot]