prusa3d/PrusaSlicer #1494: Weird slicing near the top of a spiral-mode print.

ManuelMcLure, 2018-12-23:
### Version
1.41.2+win64

### Operating system type + version
Windows 10

### Behavior
* Slicing a christmas tree bauble in spiral mode results in weird artifacts near the top.
* _Steps needed to reproduce the problem_
  Slice the attached model with the attached configuration.
* _Expected Results_
Model slices correctly.
* _Actual Results_
 The last few mm of the model are weirdly sliced. The model shows as manifold, and the "fix through Netfabb" option does not change the behavior.
[image]

* _Is this a new feature request?_
No.

#### STL/Config (.ZIP) where problem occurs
[Slic3rSlicingProblem.zip](https://github.com/prusa3d/Slic3r/files/2705732/Slic3rSlicingProblem.zip)

BerndJM, 2018-12-23:
Uncheck the "Detect thin walls" option should help ...

bubnikv, 2019-01-08:
Solved with the "Detect thin walls" option, closing.

closed 2019-01-08 by bubnikv