prusa3d/PrusaSlicer #1498: slice are missing on very thin and twisted fin parts

ANAU-designs, 2018-12-24:
### Version
Slic3rPE-1.41.2
Version 1.41.2+Linux64

### Operating system type + version
Linux 4.15.0-38-generic #41-Ubuntu SMP Wed Oct 10 10:59:38 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

### Behavior
The fins in the electric jet engine provided as attachment are not correctly sliced and therefore not printed by my (not yet entirely tuned i3 MK3), as you can see some slice are missing and also in the final result... I am new to your software and process so I did not realized it before printing.
This is NOT a new feature request.

#### STL/Config (.ZIP) where problem occurs
[3phase_electric_jet_slicing_issue.zip](https://github.com/prusa3d/Slic3r/files/2706966/3phase_electric_jet_slicing_issue.zip)
[image]
[image]
[image]

BerndJM, 2018-12-24:
Checking the "Detect thin walls" option should help ...

bubnikv, 2019-01-02:
Did "Detect thin walls" help?

ANAU-designs, 2019-01-03:
Thank you so much for coming back to me that quickly, between your last 
message and today my computer has crashed and I have not been able to 
restore it... I will keep you posted as soon as I get this device up and 
running again... this is very unfortunate... and I am annoyed, I would 
like to thank you again, new coming soon hopefully.

bubnikv, 2019-01-29:
No response in a month, closing.

closed 2019-01-29 by bubnikv