prusa3d/PrusaSlicer #1500: 1.42.0-Alpha1 won't open from self contained folder

Tohaker, 2018-12-24:
### Version
1.42.0-alpha1

### Operating system type + version
Windows 10 64-bit 1803

### Behavior
When downloading the .zip from the release page, unzipping the folder and placing it a desired location (e.g. \Documents\3D_Printing\Applications\), the application will fail to load with the following error:
"Failed to load image from file "F:\Miles\Documents\resources\icons\sys_unlock.png"

To reproduce this:
1. Download the Slic3rPE-1.42.0-alpha1+win64-full-201812231738.zip from https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-alpha1?fbclid=IwAR3Kyvn17TtKGZ63MOc3kuUf_SeSSaOddNPMm59tkMv1NW85OHyKFiXHB-I
2. Unzip the file from downloads folder.
3. Copy the entire unzipped folder to a desired location outside of the downloads folder.
4. Run the slic3r.exe file inside the folder.
5. Observe the error.

When running the file in the downloads folder everything starts as normal, although no configs are auto imported from my existing Slic3r 1.41.2. I am unsure if this is a bug or the nature of where the executable is running from, but I will give it the benefit of the doubt.

[image]
[image]

Crash331, 2018-12-24:
I am having the same issue.

bubnikv, 2018-12-24:
is there anything special about your f: drive? Is it being mapped from
somewhere? Will slic3r run from another drive? Is your windows installation
somehow locked down? For example isnt it a company laptop?

Mavi222, 2018-12-24:
Same problem. When I put it in the C:\ root folder it works. If I put it anywhere else it thinks the Resources folder is like.. one folder above my actual Resources folder 

(Resources folder is in "F:\Programy\slic3r 1.4.2\resources", it tries to find Resources folder in "F:\Programy\resources"

Crash331, 2018-12-24:
Nothing is special about mine.  I tried to copy the contents of the zip file to D:\Program Files\Prusa3d\Slic3rPE-1.42.0-alpha1 and got the same error as Tohaker.

When I unzipped the contents directly to my Downloads folder, the alpha runs fine.

Crash331, 2018-12-24:
[image]

Crash331, 2018-12-24:
[image]

Tohaker, 2018-12-24:
> is there anything special about your f: drive? Is it being mapped from somewhere? Will slic3r run from another drive? Is your windows installation somehow locked down? For example isnt it a company laptop?
> […](#)

It's my own machine, and whilst the default Documents folder is mapped to F, so its default Downloads, where it runs from. Seems like a hardcoded file path but havent had the time to dive into the code myself.

Crash331, 2018-12-24:
It also runs fine from "C:\Slic3rPE-1.42.0-alpha1"

Mavi222, 2018-12-24:
The problem is in the space, yes. I just noticed that after a bit of testing. If the path to the executable has space, it totally ignores the full path and it fails to load.

Crash331, 2018-12-24:
> edit: and I am missing all the default presets in this new version

I had to run with Wizard again and it loaded the filament and printer presets for me.

Mavi222, 2018-12-24:
> I had to run with Wizard again and it loaded the filament and printer presets for me.
Oh I totally forgot about the wizard. I thought it popped up automatically on the first start. I got all my presets back after running the wizard, thanks!

Tohaker, 2018-12-24:
Building from source seems a bit challenging seeing as they've removed Perl in this update. The instructions are very old too, may just have to wait until they release new instructions.

varukasalt, 2018-12-24:
Same issue. Moved everything to Users folder and it works now.

BAKMS, 2018-12-27:
1.42.0a1 will only launch when the program is copied to my desktop of Windows 10 (1803). If i move the program anywhere else on my C: or other drives is fails to open.

mgg4, 2018-12-28:
Opening with the slic3r-console.exe executable seems to work.

PatB42, 2019-01-02:
trying all option , read here , move on C; , desktop , copy paste....................not one work still same can't open slic3r !!

bubnikv, 2019-01-03:
We will release another alpha tomorrow.

bubnikv, 2019-01-03:
Fixed with 0d7e53badb4c62ff1a8638ca916109b524f52b6c
by adding some missing UTF8 to wxString path explicit conversions.

closed 2019-01-03 by bubnikv

angelshce22, 2019-04-01:
Im having the same problem For Slic3rPE-1.42.0-beta1 And nothing works

bubnikv, 2019-04-02:
@angelshce22 what problem? What operating system? What happens?

angelshce22, 2019-04-02:
@bubnikv windows 8.1 , when trying to install the new version i see this error

https://gyazo.com/994fa00af1fd24b96aa1d55d840bce82

Was fixed by removing in roaming the folder of prusa

bubnikv, 2019-04-03:
it is always useful to us to get the content of the data directory for
inspection in such a case. Anyway, I am happy it works for you now.