prusa3d/PrusaSlicer #1501: Po přepnutí do češtiny program zhavaruje a nejde znovu spustit

joka-cz, 2018-12-24:
### Version
_1.42.0 Alpha_

### Operating system type + version
_Windows 10 Pro 64bit_

### Behavior
* Po přepnutí na češtinu program havaruje a pak již nejde spustit

[image]

Mavi222, 2018-12-24:
Právě jsem zkusil přepnout do češtiny a mám úplně ten stejný problém.

Zatím můžeš hodit program zase do angličtiny tím že půjdeš do %appdata%\Slic3rPE-alpha ( takže stačí zmáčknout WIN+R, zobrazí se ti spouštěcí okno, tam vlož "%appdata%\Slic3rPE-alpha" a enter. )

Pak otevři slic3r.ini v textovém editoru (třeba poznámkový blok), a změň devátý řádek na "translation_language = en_US"

bubnikv, 2019-01-02:
This should be fixed with #a7d8146. Closing.

closed 2019-01-02 by bubnikv