prusa3d/PrusaSlicer #1502: Upload to duet does not start th eprint

Patrik-svobodik, 2018-12-24:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2019-01-02:
Is it a regression issue? Did it work in 1.41? We do not use Duet Wifi, so it is a bit difficult for us to test. Thanks.

Patrik-svobodik, 2019-01-02:
It works with 1.41 so it is a problem with the new upload queue, it uploads
(sometimes it does not and says it did) and then nothing happens, the
command to start printing never gets sent and sometimes it is never
uploaded, it creates a empty file and I have to upload it manually, it is
really anoying

bubnikv, 2019-01-02:
@vojtechkral will look into it.

vojtechkral, 2019-01-03:
@Patrik-svobodik Thanks for the report. The problem was due to code that was removed erroneously during refactoring.

Tomorrow there will be another alpha release with a fix for this issue (as well as others).

vojtechkral, 2019-01-09:
@Patrik-svobodik Did you get the chance to test this with the alpha2? Does it work now? Thanks!

Patrik-svobodik, 2019-01-09:
It works now, but sometimes it does not upload at all, it says uploaded but
it just generates a blank file

bubnikv, 2019-01-10:
That means, if you are willing to help, please capture the TCPIP log, so we may deduct what went wrong.

Patrik-svobodik, 2019-01-11:
@bubnikv hey, how should I do it? My main problem is that it shows done but it is not

Patrik-svobodik, 2019-01-11:
@vojtechkral I just tried it, it says done but it is still not uploaded, I have to wait and then in will start the print, the indicator should tell you how much it has uploaded but it does not

Kriechi, 2019-02-24:
The upload+start works for me with 1.42.0-alpha5+, but the progress bar / upload indicator is useless. It jumps between 0% and 100% (Completed).

commit 30184cc0 (2019-01-04): Fix #1502 Upload to duet does not start the print
Code was erroneously removed during refactoring
closed 2026-09-29 by prusaslicer-bot[bot]