prusa3d/PrusaSlicer #1503: Strange behavior while brining curved perimeters

uepsie, 2018-12-24:
### Version
1.42-alpha

### Operating system type + version
Windows 10 (1809) 64Bit

### Behavior
When printing curved perimeters, the mk3 (FW 3.5) slows down extremely and seems to be stuttering. Linear motion is fast. 

Using default pla profile and all default settings

rmoravcik, 2018-12-26:
@uepsie Is this what you see? I was trying to print some simple objects with new alpha version and I can see some strange artifacts of curved perimeters too.

Here is and example of object (https://www.thingiverse.com/thing:40487) printed using the same settings, same filament, same printer and sliced using the last stable version of Slic3r PE (1.41.2) and new alpha (1.42):

[image]

And here are generated gcodes:
[Safety_Guard_gcodes.zip](https://github.com/prusa3d/Slic3r/files/2710001/Safety_Guard_gcodes.zip)

uepsie, 2018-12-26:
another thing I noticed it the gcode file is 4 times as big as the one sliced with 1.41 and same settings

dumski, 2019-01-02:
I've noticed that problem appears also in earlier version. In 1.41.2 stable Slic3r is generating lots of movements in gap fills. BTW, I attached generated gcode from this version, can somebody print it on dry-run mode, to see if this happening on some other printers? I have delta and CoreXY, both driven by Mega2560.

[pulley10mm-1.41.2.gcode.txt](https://github.com/prusa3d/Slic3r/files/2720159/pulley10mm-1.41.2.gcode.txt)
[pulley10mm-1.42alpha.gcode.gcode.txt](https://github.com/prusa3d/Slic3r/files/2720160/pulley10mm-1.42alpha.gcode.gcode.txt)