prusa3d/PrusaSlicer #1504: Search box for settings

lf-, 2018-12-24:
### Version
_Version of Slic3r used goes here_
1.41.2

### Behavior
* _Describe the problem_
One feature I really miss from Cura is its search for settings as it's sometimes hard to figure out where some settings are. For instance, if retraction distance is in the printer extruder settings, in the print settings, etc (for the record, it's in the extruder settings).

As it is 2018, this box really should have fuzzy matching. When the user enters a search and clicks on the setting they want to see, it would be reasonable to either jump to the setting box in question, placing the cursor in it, or expose the settings directly in the dialog.

_Is this a new feature request?_
Yes

neslekkim, 2019-08-25:
I would like this also, often looking for settings, and cannot find them, either because they are named differently than other slicers (it could search the tooltip text and setting name), or because they are put under advanced or something.

men8ifr, 2020-04-20:
I would like this also - why was this request closed?

lukasmatena, 2020-04-20:
This issue is open, you can see that just below its title. It was never closed.

The search function will in fact make it into 2.3 release.

men8ifr, 2020-04-26:
Ooh excellent I will look forward to it

bubnikv, 2020-10-16:
A fuzzy search dialog will be in PrusaSlicer 2.3.0-alpha1. Closing.

closed 2020-10-16 by bubnikv