prusa3d/PrusaSlicer #1505: Moving to origin after completing a single objects layer

fabi280, 2018-12-24:
### Version
Version 1.42.0-alpha1+win64

### Operating system type + version
Windows 10, Edu, 64bit

### Behavior
After adding 2 different stl to one buildplate and arranging them, the output gcode contains a movement to a negative coordinate in the X and Y direction after finishing each object.
For example, after a layerchange
```
;LAYER:40
G1 Z8.240 F7800.000
; PURGING FINISHED
G1 E0.00101 F4800.00000
G92 E0
**G1 X-38.500 Y-50.500 F7800.000**
G1 X132.424 Y112.832
```
Here is a screenshot from a resulting gcode in the Octoprint GCode Viewer
[image]

This behaviour doesn't exist on the Version 1.41.2+win64 Version

bubnikv, 2018-12-24:
Please attach a 3MF file with which we can reproduce the issue. Thanks.

fabi280, 2018-12-25:
[Bugreport.zip](https://github.com/prusa3d/Slic3r/files/2707845/Bugreport.zip)
I've attached the 2 example STL files, the resulting 3MF and the configuration

jrdiaz, 2018-12-25:
It if helps, the problem in present in OSX version too

[image]

Rowbotronics, 2019-01-05:
I'm having this issue as well. If I slice a single object there will be only a few moves off into the boonies, but when I add multiple parts the issue gets much worse. Un-checking "Avoid crossing perimiters" eliminates the issue, and drastically reduces slice time.

bubnikv, 2019-01-07:
Fixed with 026bc801e33778ebc5500ad0a3aac9437bca181f
Closing

closed 2019-01-07 by bubnikv