prusa3d/PrusaSlicer #1506: Segfault moving support blocker

ctrlbru, 2018-12-24:
Version is 1.42.0-alpha1+linux64 on Mint 19.1

Steps to reproduce with any stl, default mk3 configs:

Import an stl
Define a support blocker
Go to preview, wait for processing to end
Go back to 3D editor view
Move the support blocker
Go to preview
Segmentation fault

No other output in console except the segmentation fault error. It doesn't seem to happen with support enforcer.

lukasmatena, 2018-12-25:
I managed to reproduce. It appears to only happen when the slicing is triggered by switching to preview tab (background processing must be off) and there is a need to reslice supports. It gives an impression that the canvas tries to render something that is touched by another thread, but I may be wrong. Someone else might see the problem (and the fix) rightaway from the log output and backtrace below.

Log output:
```
[2018-12-25 18:21:54.086336] [0x00007fffd9e1b700] [info]    Staring the slicing process.
[2018-12-25 18:21:54.086532] [0x00007fffd9e1b700] [info]    Support generator - Start
[2018-12-25 18:21:54.086569] [0x00007ffff7fc0400] [debug]   Loading print object toolpaths in parallel - start
[2018-12-25 18:21:54.086639] [0x00007fffd9e1b700] [info]    Support generator - Creating top contacts
[2018-12-25 18:21:54.086764] [0x00007fffd9e1b700] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2018-12-25 18:21:54.086830] [0x00007fffd9e1b700] [debug]   TriangleMesh::repair() started
[2018-12-25 18:21:54.086910] [0x00007fffd9e1b700] [trace]   	stl_check_faces_exact
[2018-12-25 18:21:54.088417] [0x00007fffd9e1b700] [trace]   	stl_fix_normal_directions
[2018-12-25 18:21:54.088502] [0x00007fffd9e1b700] [trace]   	stl_fix_normal_values
[2018-12-25 18:21:54.088571] [0x00007fffd9e1b700] [trace]   	stl_calculate_volume
[2018-12-25 18:21:54.088620] [0x00007fffd9e1b700] [trace]   	stl_verify_neighbors
[2018-12-25 18:21:54.088664] [0x00007fffd9e1b700] [debug]   TriangleMesh::repair() finished
[2018-12-25 18:21:54.088703] [0x00007fffd9e1b700] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2018-12-25 18:21:54.088774] [0x00007fffd9e1b700] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2018-12-25 18:21:54.088848] [0x00007fffd9e1b700] [debug]   TriangleMeshSlicer::slice
[2018-12-25 18:21:54.088890] [0x00007fffd9e1b700] [debug]   TriangleMeshSlicer::_slice_do
[2018-12-25 18:21:54.090791] [0x00007fffd9e1b700] [debug]   TriangleMeshSlicer::_make_loops_do
[2018-12-25 18:21:54.091355] [0x00007fffd9e1b700] [debug]   TriangleMeshSlicer::slice finished
[2018-12-25 18:21:54.091373] [0x00007fffd9e1b700] [debug]   TriangleMeshSlicer::make_expolygons in parallel - start

Thread 20 "slic3r" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd8db9700 (LWP 19757)]
0x0000000000c56f64 in Slic3r::GUI::GLCanvas3D::_load_print_object_toolpaths(Slic3r::PrintObject const&, std::vector<std::string, std::allocator<std::string> > const&)::{lambda(tbb::blocked_range<unsigned long> const&)#3}::operator()(tbb::blocked_range<unsigned long> const&) const ()
```

Full stack:
```
(gdb) bt
#0  0x0000000000c56f64 in Slic3r::GUI::GLCanvas3D::_load_print_object_toolpaths(Slic3r::PrintObject const&, std::vector<std::string, std::allocator<std::string> > const&)::{lambda(tbb::blocked_range<unsigned long> const&)#3}::operator()(tbb::blocked_range<unsigned long> const&) const ()
#1  0x0000000000c5e786 in tbb::interface9::internal::start_for<tbb::blocked_range<unsigned long>, Slic3r::GUI::GLCanvas3D::_load_print_object_toolpaths(Slic3r::PrintObject const&, std::vector<std::string, std::allocator<std::string> > const&)::{lambda(tbb::blocked_range<unsigned long> const&)#3}, tbb::auto_partitioner const>::execute() ()
#2  0x000000000146f1cd in tbb::internal::custom_scheduler<tbb::internal::IntelSchedulerTraits>::local_wait_for_all(tbb::task&, tbb::task*) ()
#3  0x00000000014655e6 in tbb::internal::arena::process(tbb::internal::generic_scheduler&) ()
#4  0x0000000001472614 in tbb::internal::market::process(rml::job&) ()
#5  0x00000000014741d6 in tbb::internal::rml::private_worker::run() ()
#6  0x0000000001474469 in tbb::internal::rml::private_worker::thread_routine(void*) ()
#7  0x00007ffff3e5d6ba in start_thread (arg=0x7fffd8db9700) at pthread_create.c:333
#8  0x00007ffff331341d in clone () at **../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

bubnikv, 2019-01-09:
Fixed with 713ec1f06549315c198fff787880321198028e59 and a476e4ca0f88d31c5307d4f1c921e149004fdf63

There was indeed an unsafe access to already invalidated support layers from the G-code preview UI.

closed 2019-01-09 by bubnikv