prusa3d/PrusaSlicer #1507: About dialog:  assert "Assert failure" failed in EndModal(): either wxDialog:EndModal called twice or ShowModal wasn't called

mkyral, 2018-12-24:
### Version
1.42.0-alpha1

### Operating system type + version
Gentoo Linux

### Behavior
Open About dialog and click inside (try copy Slic3r version or close the dialog).

This error is shown:
[image]

ASSERT INFO:
/var/tmp/portage/x11-libs/wxGTK-3.0.4/work/wxWidgets-3.0.4/src/gtk/dialog.cpp(199): assert "Assert failure" failed in EndModal(): either wxDialog:EndModal called twice or ShowModal wasn't called

BACKTRACE:
[1] wxDialog::EndModal(int)
[2] Slic3r::GUI::AboutDialog::onCloseDialog(wxEvent&)
[3] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[4] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[5] wxEvtHandler::TryHereOnly(wxEvent&)
[6] wxEvtHandler::ProcessEventLocally(wxEvent&)
[7] wxEvtHandler::ProcessEvent(wxEvent&)
[8] wxWindowBase::TryAfter(wxEvent&)
[9] wxDialogBase::EmulateButtonClickIfPresent(int)
[10] wxDialogBase::OnCloseWindow(wxCloseEvent&)
[11] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[12] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[13] wxEvtHandler::TryHereOnly(wxEvent&)
[14] wxEvtHandler::ProcessEventLocally(wxEvent&)
[15] wxEvtHandler::ProcessEvent(wxEvent&)
[16] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[17] wxWindowBase::Close(bool)
[18] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[19] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[20] wxEvtHandler::TryHereOnly(wxEvent&)
[21] wxEvtHandler::ProcessEventLocally(wxEvent&)
[22] wxEvtHandler::ProcessEvent(wxEvent&)
[23] wxWindowBase::TryAfter(wxEvent&)
[24] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[25] g_closure_invoke
[26] g_signal_emit_valist
[27] g_signal_emit
[28] g_closure_invoke
[29] g_signal_emit_valist
[30] g_signal_emit
[31] g_closure_invoke
[32] g_signal_emit_valist
[33] g_signal_emit
[34] gtk_propagate_event
[35] gtk_main_do_event
[36] g_main_context_dispatch
[37] g_main_loop_run
[38] gtk_main
[39] wxGUIEventLoop::DoRun()
[40] wxEventLoopBase::Run()
[41] wxDialog::ShowModal()
[42] Slic3r::GUI::about()
[43] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[44] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[45] wxEvtHandler::TryHereOnly(wxEvent&)
[46] wxEvtHandler::ProcessEventLocally(wxEvent&)
[47] wxEvtHandler::ProcessEvent(wxEvent&)
[48] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[49] wxMenuBase::SendEvent(int, int)
[50] g_closure_invoke
[51] g_signal_emit_valist
[52] g_signal_emit
[53] gtk_widget_activate
[54] gtk_menu_shell_activate_item
[55] g_closure_invoke
[56] g_signal_emit_valist
[57] g_signal_emit
[58] gtk_propagate_event
[59] gtk_main_do_event
[60] g_main_context_dispatch
[61] g_main_loop_run
[62] gtk_main
[63] wxGUIEventLoop::DoRun()
[64] wxEventLoopBase::Run()
[65] wxAppConsoleBase::MainLoop()
[66] wxEntry(int&, wchar_t**)
[67] main
[68] __libc_start_main
[69] _start

lukasmatena, 2018-12-24:
Thanks for the report. You're using wxWidgets 3.0.4, the crash seems not to occur with 3.1.
However, it is probably still a bug in Slic3r. wxDialog::Destroy should not be invoked for stack allocated wxDialog. I hopefully fixed it with https://github.com/prusa3d/Slic3r/commit/9bd57285081e613c05988173a4f369bfac80b05c.

I better leave the issue open for now, will close it later.

bubnikv, 2019-01-04:
fixed, closing.

closed 2019-01-04 by bubnikv

mkyral, 2019-01-04:
Well, I'm not sure about it ;-)

[image]

I'm using 3.0.4 because it is latest stable release. Version 3.1.x is development only and not available in Gentoo portage.

bubnikv, 2019-01-05:
Would you please attach the backtrace? Just click the backtrace button at
the assert screen.

We cannot support old wxWidgets. I am sorry that your distro's wxWidgets
are not up to date.

mkyral, 2019-01-05:
Sure
```
ASSERT INFO:
/var/tmp/portage/x11-libs/wxGTK-3.0.4/work/wxWidgets-3.0.4/src/gtk/dialog.cpp(199): assert "Assert failure" failed in EndModal(): either wxDialog:EndModal called twice or ShowModal wasn't called

BACKTRACE:
[1] wxDialog::EndModal(int)
[2] Slic3r::GUI::AboutDialog::onCloseDialog(wxEvent&)
[3] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[4] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[5] wxEvtHandler::TryHereOnly(wxEvent&)
[6] wxEvtHandler::ProcessEventLocally(wxEvent&)
[7] wxEvtHandler::ProcessEvent(wxEvent&)
[8] wxWindowBase::TryAfter(wxEvent&)
[9] wxDialogBase::EmulateButtonClickIfPresent(int)
[10] wxDialogBase::OnCloseWindow(wxCloseEvent&)
[11] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[12] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[13] wxEvtHandler::TryHereOnly(wxEvent&)
[14] wxEvtHandler::ProcessEventLocally(wxEvent&)
[15] wxEvtHandler::ProcessEvent(wxEvent&)
[16] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[17] wxWindowBase::Close(bool)
[18] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[19] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[20] wxEvtHandler::TryHereOnly(wxEvent&)
[21] wxEvtHandler::ProcessEventLocally(wxEvent&)
[22] wxEvtHandler::ProcessEvent(wxEvent&)
[23] wxWindowBase::TryAfter(wxEvent&)
[24] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[25] g_closure_invoke
[26] g_signal_emit_valist
[27] g_signal_emit
[28] g_closure_invoke
[29] g_signal_emit_valist
[30] g_signal_emit
[31] g_closure_invoke
[32] g_signal_emit_valist
[33] g_signal_emit
[34] gtk_propagate_event
[35] gtk_main_do_event
[36] g_main_context_dispatch
[37] g_main_loop_run
[38] gtk_main
[39] wxGUIEventLoop::DoRun()
[40] wxEventLoopBase::Run()
[41] wxDialog::ShowModal()
[42] Slic3r::GUI::about()
[43] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[44] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[45] wxEvtHandler::TryHereOnly(wxEvent&)
[46] wxEvtHandler::ProcessEventLocally(wxEvent&)
[47] wxEvtHandler::ProcessEvent(wxEvent&)
[48] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[49] wxMenuBase::SendEvent(int, int)
[50] g_closure_invoke
[51] g_signal_emit_valist
[52] g_signal_emit
[53] gtk_widget_activate
[54] gtk_menu_shell_activate_item
[55] g_closure_invoke
[56] g_signal_emit_valist
[57] g_signal_emit
[58] gtk_propagate_event
[59] gtk_main_do_event
[60] g_main_context_dispatch
[61] g_main_loop_run
[62] gtk_main
[63] wxGUIEventLoop::DoRun()
[64] wxEventLoopBase::Run()
[65] wxAppConsoleBase::MainLoop()
[66] wxEntry(int&, wchar_t**)
[67] main
[68] __libc_start_main
[69] _start
```

And if you don't support official latest release of wxwidgets, you should specify minimal version in cmake ;-)