prusa3d/PrusaSlicer #1510: Config bundle not being importet correctly [1.42.0-alpha1]

Boby71, 2018-12-25:
### Version
Slic3r 1.42.0-alpha1

### Operating system type + version
Win64

### Behavior
I tried the new alpha version 1.42.0 and exported the config bundle from the previous version (1.41.2).
Unfortunately, most of my user settings are gone now (e.g. retraction settings on the printer/extruder tab or my filament settings [e.g. user-defined settings]).

Boby71, 2018-12-25:
Man, that's bad...when I now import my config bundle into the old version, my settings are also gone. Don't know what happend, but I forgot to take a configuration snapshot and now I can set up all parameters again...:-(

bubnikv, 2019-01-02:
Please provide your config bundle, so we can analyze it. Thanks.

bubnikv, 2019-01-08:
You provided the config directory for 1.41. The 1.42 alpha stores its data into Slic3rPE-alpha directory to not mess up the production Slic3r.

Boby71, 2019-01-26:
This issue has been fixed already

closed 2019-01-26 by Boby71