prusa3d/PrusaSlicer #1511: System presets listed among user presets (and can't be deleted) [1.42.0-alpha1]

Boby71, 2018-12-25:
### Version
1.42.0-alpha1

### Operating system type + version
Win64

### Behavior
The sytem presets of Taulman T-Glase, Verbatim BVOH and Verbatim PP are doubled in the filament settings and being showed among the user presets as well. But they can't get deleted from the user presets because the software thinks, I want to delete the system setting.

Remark: I imported a config bundle vom 1.41.2 before - maybe this confused the alpha version?

[image]

Boby71, 2018-12-25:
After some adding/deleting of user defined filament presets, the list got shorter:
[image]

bubnikv, 2018-12-25:
This issue and the previous one is likely caused by the previous slic3r
exporting the system presets into the config bundle, and then loading those
system presets and storing them into user's presets. We have suppressed
export of the system presets into the config bundle in 1.42.

danielegger1, 2018-12-28:
To delete inside Slic3r was not possible to I exported the config bundle, deleted the 2 wrong filament types, initialized slic3r with default settings and reimported it.

bubnikv, 2019-01-03:
Fixed with b1a7ac07c93fce30fd90054e2f7cb2d2daa2f65e

    
    When loading a user provided Config Bundle (not a vendor Config Bundle),
    profiles of the same name as the system profiles already loaded into Slic3r
    will be ignored, while emitting an error message into the log.
    
    Also when loading a user provided Config Bundle, imported vs. overwritten
    profiles are reported into the log as well.

closed 2019-01-03 by bubnikv