prusa3d/PrusaSlicer #1512: Printer Host Upload Queue vanishes in the background [1.42.0-alpha1]

Boby71, 2018-12-25:
### Version
1.42.0-alpha1

### Operating system type + version
Win64

### Behavior
When trying to display the host upload queue, it shows up the first time for a few seconds - but then it vanishes into the background and when being selected again, it's vanishing again. Not really able to bring the window in the foreground.

Boby71, 2018-12-25:
Here's a video of the behaviour:
[Capture_1.zip](https://github.com/prusa3d/Slic3r/files/2708994/Capture_1.zip)
The first time, I had to get the queue from my second screen which was not part of the recording.
But after a few seconds it vanishes and steadily went behind the slicer main window.

bubnikv, 2018-12-25:
This behavior is caused by the Slic3r's 3D window grabbing the mouse focus
even if the main Slic3r's window is not active. This is an old annoyance.

bubnikv, 2019-01-02:
Fixed with b7a92b5525c63be6d99879b8d3a941c1242a7633

The canvas3d used to grab focus when mouse entered a window even if the top level window was not active. This was now changed and the top level must be active for the canvas3d to grab the focus.

Closing.

closed 2019-01-02 by bubnikv