prusa3d/PrusaSlicer #1513: Cannot move part in Z direction

faboaic, 2018-12-25:
### Version
Slic3rPE-1.42.0-alpha1+win64-full-201812231738

### Operating system type + version
Win10 x64

### Behavior

* _Describe the problem_

I can move part in X and Y direction with the new function.
If I move it in Z direction up or down it instantly goes back to default position.

* _Steps needed to reproduce the problem_

Use move symbol in x , y and z direction.

* _Expected Results_

Z works like X and Y. 
I would need to lower the part below the bed (like with not-so-comfortable cut function) so that lower part is not printed.
Or need to lift a part up in the air if you want to continue a failed print... or just do some tests with support function...

* _Actual Results_

Does not work...

  * _Screenshots from __*Slic3r*__ preview are preferred_

[image]

[image]

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

[cannotmoveZ.zip](https://github.com/prusa3d/Slic3r/files/2708642/cannotmoveZ.zip)

EXOgreen, 2018-12-25:
I have found that this only affects files that are not parts of a "part".  With the new split to object and split to part options, a file with two pieces split into objects will both always be on the ground. But with a file that is split into a part, both items will stay where they are and can be moved in the z direction. Would it be possible to at least allow us to add other files to a group AKA part? 

I would suggest a two part fix
1. Adding a "group to part" UI icon next to the two splits.
2. for individual objects, graying out the Z position box and adding hover text that says that its only available for grouped parts.

With this also allowing users to group/ungroup parts by dragging them in the sidebar into/out of parts would be 👌

faboaic, 2018-12-26:
EXOgreen ... sounds complicated.
Cannot we just take a part and lift or lower it to our desired Z position and fine without any cut/split/group/append/modify/... ?
It has been months that I moved from Cura to Slic3r PE (still happy with slic3r PE :-) ) but I think I remember that in cura you just took the part and lifted it up and that was it.

bubnikv, 2018-12-26:
You should not need to move the object up in Z. If you are printing without
raft, you print the first object layer on the print bed. If you are
printing with raft, then Slic3r will apply the prescribed number of raft
layers for you, thus it will lift the object as required.

EXOgreen, 2018-12-26:
I would disagree, I have a model that can/should be stacked on top of each other. But they are individual files. If I bring them all into Slic3r, I cannot stack them, and cannot add them together as one part so they all are affected by one change, ex rotation or scale.

faboaic, 2018-12-26:
@bubnikv There are many cases when moving the part in Z direction really makes a lot of sense.
See expected result in my first post.

Try to print a part that has one area of its underside touching the build plate and another area of its underside 0.5mm higher in Z direction. 
You will get better results if you lift it up so that support is everywhere and not only 0.5mm below that higher area. 
0.5mm is too less height for interface layers, etc... it will be really bad result without lifting it up.
.... just as a not so usual example ;-)

But to resume a failed print is really easy with lifting the part up. 

I cannot find a reason why it should be forbidden to move a part up or down.
With the "lay flat" icon you can drop it down on the bed again, if necessary.

bubnikv, 2019-01-02:
> I would need to lower the part below the bed (like with not-so-comfortable cut function) so that lower part is not printed.

What is not-so-comfortable on the cut function?

> But to resume a failed print is really easy with lifting the part up.

I don't understand how this is supposed to help to resume a failed print.

> I cannot find a reason why it should be forbidden to move a part up or down.

Any functionality, which is not needed, will stay in somebody's way.

> Try to print a part that has one area of its underside touching the build plate and another area of its underside 0.5mm higher in Z direction.
You will get better results if you lift it up so that support is everywhere and not only 0.5mm below that higher area.

This is easily achieved by enabling the raft.

@EXOgreen You can stack objects in Slic3r by right clicking on the object in the 3D view or at the object list, and selecting "Add Part -> Load ...". You can then manipulate the parts in Z as you wish.

EXOgreen, 2019-01-02:
Can we add this functionality of combining files as parts in the sidebar for files already loaded in Slic3r. This is what's happening with right click > load, but not for existing parts. So if someone has loaded two file's and made changes they can be combined into a part, essentially grouping them.

bubnikv, 2019-01-02:
Yes, this makes sense, possibly with drag & drop at the side bar.

faboaic, 2019-01-04:
> What is not-so-comfortable on the cut function?

It works but is not as intuitive as just lowering the part below the Z = 0 plane.
 
> > But to resume a failed print is really easy with lifting the part up.
> 
> I don't understand how this is supposed to help to resume a failed print.

Find out the height of the failure (caliper, log, ... ) then cut the part at that height in the slicer.
Raise the part to that height in slicer. 
Export gcode, print --> Printer will continue at the layer where the failure happened.
 
> > I cannot find a reason why it should be forbidden to move a part up or down.
> 
> Any functionality, which is not needed, will stay in somebody's way.

It is needed ;-) see above.

> This is easily achieved by enabling the raft.

Maybe this was not my best example without a drawing. 
Have you ever used cura? It is very intuitive to just take a part in 3D space and rotate it in every direction and move it in every direction. Even Z ;-)

faboaic, 2019-01-04:
It is confusing for me that a part is treated differently than a modifier at moving. 

When I have a part and a modifier,  I can lift the modifier. 
Then I change the modifier to a part...  It is lifted. 
I change the lower part to a modifier,  then both parts are moved....  That is really strange behavior. 

If parts and modifiers could be positioned freely it would be feeling just normally and intuitive. 
Changing type from part to modifier and back should not move the part. 

If you need a part touching the build plate you have the perfectly working new function 'lay flat '.

mgg4, 2019-03-27:
I have also commented on another issue thread.  Stacking parts to increase productivity when producing a large number of smaller parts is an important feature.  The parts are stacked with support material turned on between the layers.  Other slicers have this capability.  The attached screen grab is from Simplify3D.
[image]

tdub415, 2019-05-07:
Another good reason to have the abiltiy to move an object above the build plate so no part is resting, is for miniature printing.  You can often get a much better angle to print it at by not having it lay flat, and instead entirely rest on supports.  The inability to manipulate models in the Z direction is about the only thing keeping me using simplify3D or Cura over Slic3r PE.

bubnikv, 2019-05-08:
> You can often get a much better angle to print it at by not having it lay flat, and instead entirely rest on supports.

Just enable the raft.

Trellian, 2019-08-14:
@bubnikv Then why not simply remove the Z Arrow and text box, since they not only do nothing, they are extremely confusing, as witnessed by the presence of this thread.  

That said, I would rather have the option to move things in the Z-axis. People make clever plans when they have options, and the nature of this community is those kinds of people. Perhaps just include a visual indicator or warning if a part is not flat on the bed?

titaniumschmapple, 2019-08-17:
Using a torus (donut) shape with 3D printing won't work properly if you can't raise the model off the build plate - the edge that touches the build plate gets flattened - so it's **VITAL** that users can lift their models off the build plate!!

More user choice is ALWAYS good. Especially if it's disabled by default and if anyone wants to use it, they simply go to the Expert settings.

bubnikv, 2019-08-19:
> Using a torus (donut) shape with 3D printing won't work properly if you can't raise the model off the build plate - the edge that touches the build plate gets flattened - so it's VITAL that users can lift their models off the build plate!!

How are you printing the object then? On raft? Without raft but with supports?

titaniumschmapple, 2019-08-19:
I can't print it successfully, even with a Raft it does the same thing!

wesparish, 2019-08-23:
I would also like this feature. For instance, I'm currently trying to stack 4mm thick fan guards so I'm not having to clear the build plate every 1.5 hours. Unfortunately, I'm not able to get a small "raft" to print between the parts. I've tried support enforcers (using 'load' to load another instance of the object marked as a support enforcer) and support blockers (by modeling the inverse of the print and extruding it very high, through the full stack) both. I've also tried just a "slab" between the parts as a support enforcer. No matter what, the slicer seems to want to build supports for supports, all the way down to the build plate, even when nothing is above the area.

[image]
[image]

with a support blocker
[image]
[image]
[image]

JohnOCFII, 2019-09-06:
I too, wish that I could raise and lower objects.  My initial use case is manually stacking items, often just as a preview of what things would look like, without having to open STLs in TinderCAD or similar.   Simplify3D has this feature. 

As others have mentioned, it could cause issues for beginners, but we do have simple, advanced, and export modes now.  It could be a preference that only shows up for other than simple mode.

Naugrimohtar, 2019-09-17:
I also want the ability to lift in z-axis. I often ‘kitbash’ where I add two files together in slicer. Imagine a had a model of a D&D character and another model of sun glasses, I can add the sun glasses, scale and move where I want. The object would merge into a single object to be printed.

Another common thing is for me to have a flat textured ‘tile’ for D&D. I want to add a wall so I want to put it ‘on top’ of the floor tile to make one piece.

hunagyp, 2019-09-20:
I would also join the "why this useful feature not available" chain.
Why? Is it hard to implement in the code?
What if I just want to generate gcode for such a model, which is "flying" in the air?
Why do you constrain the ability to move the part in 3D space anywhere the user wants?
Craftware, cura, S3D (just the ones I use) all has this very simple function - only Prusa thinks it is not useful? :D How is that?

cotestatnt, 2019-10-03:
I agree with the majority of users! Please allow the Z to be moved as everyone would like.
This "automatic" z placing of models is completely unusefull as my opinion.

kulberda, 2019-10-09:
I've been struggling with trying to figure this out for 20 minutes now.  It makes no sense.  I have two objects, one that should sit on top of the other.  The base object is effectively a raft with a custom shape and the top object is a more complex design to be printed on top of the base object.  It has a bunch of separate pieces so splitting is would be cumbersome (and non-intuitive).

Yes, I could bring both objects into another 3D software and combine them but that should be an extra uneccessary step.  I also want to be able to easily use the base with different top object designed and just swap them out.  Just let me move the thing up please!

neophyl, 2019-10-09:
Kulberda, place your base object normally, then right click on it and 'add part', select your model that you want to place on top.  You should now be able to move the 'top' part around freely including into the air on top of it.  By adding it as part of the original model Prusa Slicer groups them as a logical entity and as long as some part of the logical group touches the build plate its requirements are met.

kulberda, 2019-10-09:
Yes, thanks, that works!  But, still, not intuitive and not at all clear why that's any different than trying to reposition two separate parts on top of each other.  At the very least it should tell you why you can't move the other part up (and what to do about it).

wavexx, 2019-10-31:
Floating separate objects could still be useful to create custom supports which need to be printed with a different layer height: #3117

ProfEngr, 2019-12-23:
Over a year worth of updates to the slicer and this is still a problem? Why do the other 'big' slicers have this and PS doesn't?

darraghbr, 2019-12-23:
I ran into a usage case for this feature a month or so ago, I wanted to print some pin badges, where the pin is encapsulated in the print itself. I developed a methodology that works and can be seen on my [blog post here.](https://darraghbr.blogspot.com/2019/11/printing-pin-badges-on-prusa-i3-mk3s.html) I ended up using a huge number of raft layers which sort of worked, however the pin section was difficult to fit because the raft layers so you have to push the pin through the raft layers. 

If I was able to raise the print above the print bed and then generate supports as normal this sort of print would be much simpler.

EXOgreen, 2019-12-23:
While I do see the point of having the ability to move parts in the Z dirrection, most people I work with who use our prusas (students) dont know half what they should. I would rather the parts auto snap to the ground in almost every case.

@darraghbr There is a much simpler way to move parts in the Z direction. when two STL objects are part of a group, they will be able to move in the Z direction. there SHOULD be a way to group objects to a part but I see that hasn't been added yet. my current workaround is to add my objects to the buildplate, export the buildplate as an STL, then clear the board and import the new STL. As soon as you select split to part, you will be able to select each object and give them their own Z height. 

I would like to take this opportunity to reiterate my suggestion from when this first became an issue almost exactly a year ago ( @bubnikv ) 

I would suggest a two-part fix

    Adding a "group to part" UI icon next to the two split options.
    for individual objects, graying out the Z position box and adding hover text that says that its only available for grouped parts.

With this also allowing users to group/ungroup parts by dragging them in the sidebar into/out of parts would be 👌

darraghbr, 2019-12-24:
@EXOgreen I attempted to replicate what you have suggested and while I would consider it more of a workaround than a real solution. For instance, what if I only want to print one part? Plus your suggestion only works when one component remains on the build plate while the other one moves in Z relative to that, as soon as you move the second component off the build plate it gets pushed down in Z again. 

If you are concerned with making the software overly complex for new users, I would suggest that this feature would only be enabled if the user were using the 'Advanced' or 'Expert' mode.

Heipetri, 2019-12-29:
This seems to been problem for long time... 😮 

Well i was trying to make MMU print and because i cant z those parts it will trying to print different color in same place after slicing....

[image]

[image]

Maybe we will get solution for this in year 2020 😄 
And yeah.. i think "Enable z" for expert/advanced mode would be nice

AbuMaia, 2020-01-07:
"While I do see the point of having the ability to move parts in the Z dirrection, most people I work with who use our prusas (students) dont know half what they should. I would rather the parts auto snap to the ground in almost every case."

Most of us here are likely NOT your students. So why restrict us from being able to do something because you have a poor opinion of your students' abilities?

terribleperson, 2020-01-08:
I was looking to move a part up so I could print it on top of a (non-printed) object sitting on the bed. This should be a feature. I agree that it's undesirable to ruin prints accidentally by moving them away from the bed, but this is easily solved. Simply have a setting for "snap parts to bed" that defaults to on and is only visible in expert mode.

Strat00s, 2020-01-30:
> I was looking to move a part up so I could print it on top of a (non-printed) object sitting on the bed. This should be a feature. I agree that it's undesirable to ruin prints accidentally by moving [...]

This shouldn't be a feature, this should be a basic function...

adaycj, 2020-02-27:
An hour of my life I can't get back. All I wanted was to stack my temperature tower tests so I could edit it as needed by material without a gcode revamp. Sooo many posts about why. Why? Because the functionality is right there and the text box reverts the number I type with no explanation. Why? Because any and every object 3d manipulation software on the planet allows it, even my kids Minecraft. Years later and we are still typing paragraphs about use cases. I don't program on this project, but on the projects I do work on this would be labeled as a BUG and treated as such.  And don't even get me started on a design standard that overwrites a user input with no notification ... good grief.

zerg960, 2020-03-10:
The z-snap to build plate made me sad today. I had a large 24" x 22" object I wanted to print on my MK3S MMU2S - I had the part split into multiple interlocking pieces. It had letters on top that I wanted to be in a different color from the rest of the part, for labeling.

The STL has over 250 shells. The UI feature to bulk-set extruders took over an hour to discover.

Splitting the STL into parts meant I can't use marquee selection to disable printing on the parts outside of the build volume, or easily identify which parts I should assign to which extruder. Splitting the STL into objects meant the letters automatically snapped to the build plate - if I hadn't used the layer by layer preview, I'd have been even sadder.

The current UI is unusable for the use case of "i have 250+ shells.".

soswow, 2020-04-04:
Why can't there be a setting for that? I agree that default should be to drop. But why not give users a way to choose if he has special needs?

CrazyJake88, 2020-04-07:
I'm seeing a number of groups making face shields trying to get quick and efficient prints that need minimal human contact and stacking on the Z axis is what many are asking for. Can 'snap to z-axis' be made an advanced toggle feature so face shields can be stacked?

soswow, 2020-04-07:
@CrazyJake88 Pretty much I came back to this topic, yes. I had to switch to Cura for that. It's a shame.

piotreknow, 2020-04-07:
Gluing model to bed in z axis is maybe convenient, but more convenient would be checking box next to "Z" for that, maybe in advanced mode?

soswow, 2020-04-07:
Btw, someone (not me) has discovered this workaround. First prep. single model that contains empty layers. Do it with meshmixer or via CURA.
I do this with CURA:
1. Go to CURA. Turn the option "Automatically drop model to the build plate" off.
2. Add you models, align and move each up using number input for z (my model is 6mm, so I move second model at 6.2mm, third at 12.4mm etc)
3. File -> Export ... -> somename.stl

Now, import that stacked model into PrusaSlicer. Add Support Enforcer. Place it so it touches slightly your model and spans vertically between all layers. Slice! =) PrusaSlicer is happy now.

lukasmatena, 2020-04-07:
@soswow You can also simply add the models as parts of one object. But there is much less fun in that :-)

soswow, 2020-04-07:
@lukasmatena please explain. And don't forget to check and make sure it works what you suggesting (if I understood you correctly - Slicer won't allow you that.) 

Here is my original comment. I removed it because I thought I made a mistake. But I didn't I forgot important step:

Btw, someone (not me) has discovered this workaround. First prep. single model that contains empty layers. Do it with meshmixer or via CURA. I do this with CURA:
1. Go to CURA. Turn the option "Automatically drop model to the build plate" off.
2. Add your models, align and move each up using number input for z (my model is 6mm, so I move the second model at 6.2mm, third at 12.4mm etc)
3. File -> Export ... -> somename.stl

Now, import that stacked model into PrusaSlicer. Add Support Enforcer. Specify "Supports for enforcers only" Place it so it touches slightly your model and spans vertically between all layers. Slice! =) PrusaSlicer is happy now.

lukasmatena, 2020-04-07:
> please explain. And don't forget to check and make sure

1. Add your object
2. Add a part (Right-click the object -> Add part -> Load...)
3. Make sure the new part is selected (it should be)
4. Select the 'Move' manipulator from the left menu
5. Move the part wherever you want it (or use the object manipulation to enter precise values)

neophyl, 2020-04-07:
To add to Lukasmatenas instructions, 
-if you are stacking and want to leave a gap between the parts Prusa Slicer will then point out there are empty layers.  
-To get around this simply right click your Object,  and add a modifier, for example a cylinder and make it 0.5mm (or just wider than your extrusion width) and as high as your stack.  
-Then position it somewhere in the stack where it will be easy to trim with a blade.

No more empty layers and you have a stack of parts.  All of the printing requirements are met and you can happily print.

soswow, 2020-04-07:
@lukasmatena +++ You taught me something new today. That's cool. I didn't notice "Add Part" way of adding it.
@neophyl yeah, that was my point. Though I was using another workaround (adding support enforcer) where you suggest actual part. Similar thing. 

So many options. And so annoying considering the issue we are currently in.

davidkeko, 2020-04-10:
I would like to raise some parts in Z direction, and put different raft/supporters for each part. Check the images (top avobe and from bottom) of a gyroscope I want to print. If PS does not let me do, I will use TinkerCAD to create some thin "supporter layers" under the central parts, and pray that they will be easily removed...
[image]
[image]

simoonvance, 2020-04-16:
please! Please just make this a feature! For years people have asked to have it and it still isn't a thing! Literally every other slicer has these capabilities and yet still to this day.

davidkeko, 2020-04-17:
I used the "add part into model" function, with "support only in plate" and PS let me slice! See the images  (from top and a near-bottom slice):
[image]
[image]

NickJonesGlass, 2020-04-18:
> I cannot find a reason why it should be forbidden to move a part up or down.
> 
> Any functionality, which is not needed, will stay in somebody's way.

@bubnikv has a bigger picture view of necessary functionality.  I am not an expert.  I am just a PrusaSlicer user.  I registered on GitHub to make my first post because I would have found the requested _simple_ and _easy to implement_ flexibility in Z be very useful today.  I needed to join two thirdparty .stls to print them as one part, and would have much preferred to position one part over the other in PrusaSlicer without need for additional editing.

Please reconsider enabling this.  Thanks.

zerg960, 2020-04-18:
wonder what this line's for:

https://github.com/prusa3d/PrusaSlicer/blob/be9dcf0c77d106393b08f961042ea44cd56f8745/src/slic3r/GUI/GUI_ObjectManipulation.cpp#L623

refers to something about a drop to bed button. What'd happen if we nuked that if statement and had the drop to bed button always be available? (possibly in advanced or expert mode).

davidkeko, 2020-04-18:
> >  I needed to join two thirdparty .stls to print them as one part, and would have much preferred to position one part over the other in PrusaSlicer without need for additional editing.
> 
> Please reconsider enabling this. Thanks.

Until then, you should do it outside PS. For example, Tinkercad or similar, the one you feel more confortable

NickJonesGlass, 2020-04-19:
> I needed to join two thirdparty .stls to print them as one part, and would have much preferred to position one part over the other in PrusaSlicer without need for additional editing.
>
> Please [...]

I indicated in my post that I had to do what you are suggesting.

carlosjln, 2020-05-26:
This is BULLSHIT! as a sr software developer I'll tell whoever is denying to work on this feature that YOU DO NOT MAKE CHOICES FOR THE USER! you create features as options where each user opts in on what they want to use...

For the guy with the argument that he's working with students that do not know what they're doing... let me guess, you let those students use a 3D printer which's hot end es at 200c, right? what could go wrong if they place the object 100mm above the build plate? wasted filament and that's it. Lesson learned.

This reminds me, yet again, that I'm trying to use prusaslicer and I go back to cura because it doesn't force you to go through some shitty process that someone else decided you should do.

This is simple, in expert mode, add a checkbox and label "Allow free object manipulation" or whatever is more intuitive. 

P.S. Go back to take classes about improving UX (User Experience), you should know how important that is.

simoonvance, 2020-05-26:
I mean just like put in a setting in preferences to enable this as a feature like cura, it's not incredibly difficult and you guys are getting paid to do this sort of stuff anyways, literally everyone wants it

kulberda, 2020-05-27:
Well, I'm back to this same problem again.  Although this time I have a mesh that isn't completely flat on the bottom and all I want to do is move it down through the base plate a bit.  I'm sure there's probably some other "trick" to solve this but (and I really hate it when people use all caps but I think the situation calls for it) ALL I WANT TO DO IS BE ABLE TO CHANGE THE Z POSITION!!!!!

carlosjln, 2020-05-27:
@kulberda I feel you, and I'd say I'm sorry I literally screamed on my post above, but that's what this frustration feels like. A stubborn developer that is telling us what to do and how to do things when you can see the shit load of people on this thread yelling out for that function. 

I think how social media works these days, we should come up with a hashtag and everyone spam their accounts. Something like @prusa #FreeZaxis

davidkeko, 2020-05-27:
> Well, I'm back to this same problem again. Although this time I have a mesh that isn't completely flat on the bottom and all I want to do is move it down through the base plate a bit. I'm sure [...]

I know, man. Try cutting the bottom in a 3D editor to make it completely flat.

neophyl, 2020-05-27:
> Well, I'm back to this same problem again. Although this time I have a mesh that isn't completely flat on the bottom and all I want to do is move it down through the base plate a bit. I'm sure [...]

The 'trick' for this one is to use the cut function.  Trim off the bottom of the model and it sits flat on the plater where you want it.  Theres a whole tool for it with options and everything.

Same solution for davidkeko's issue.  Btw cutting the bottom flat is very easy to do in Blender too but its easier to do that using Slicers Cut function.

kulberda, 2020-05-27:
> The 'trick' for this one is to use the cut function. Trim off the bottom of the model and it sits flat on the plater where you want it. Theres a whole tool for it with options and everything.
>
> Same [...]

I used to use the cut function all the time but I thought it was removed.  That was my first thought but I can't find it anywhere.  It's definitely not in the edit menu (right click context menu) or obviously anywhere else I'm looking.

[99 more replies]

closed 2026-09-01 by lukasmatena