prusa3d/PrusaSlicer #1514: /* Feature request */ Meshmixer like tree support for FDM printing

martinowren, 2018-12-25:
_Is this a new feature request?_
Yes

You have showed you can do it with the new Alpha SLA support. So make something like it for FDM ;)

BAKMS, 2018-12-27:
I would like that too...
Also would like to see "Optimize Orientation" on the i3 side too not just SL1

cnlson, 2018-12-29:
+1 to both although tree support is more important.

Patrik-svobodik, 2019-01-18:
+1 for both

Patrik-svobodik, 2019-01-18:
And maybe make them like cura so they are more stable, they can be “like trees”

kmshort, 2019-05-15:
Tree supports for FDM would be fantastic.
Code is already there for SLA, so how hard can it be?

Also, "painting" an area on an object that needs support, with supports being auto-generated to target that specific area. That would be really useful.

bubnikv, 2019-05-15:
> Tree supports for FDM would be fantastic.
> Code is already there for SLA, so how hard can it be?

The SLA tree supports will certainly not work for FDM without major changes.

> Also, "painting" an area on an object that needs support, with supports
being auto-generated to target that specific area. That would be really
useful.

We were thinking about that as well.

vertigo235, 2019-06-16:
I really hope the PrusaSlicer team can get back to working on FDM features soon.  Would really love to see some improvements with FDM supports and this is not limited to tree supports.

gyoussef1980, 2019-07-14:
I also would love to see tree supports on an upcoming release. I also use Cura and tree supports work quite well. I have contacted Prusa support requesting them and I sent Josef a message via reddit hope Prusa responds

WarrenSchultz, 2019-08-09:
This is possible to hack with extra steps in the latest release if you use an SL1 profile to generate the supports on the base plate, then export as STL with Supports. You can then import that into an FDM project. There is extra support material on the bed, but it works.

All that to say, Prusa has all the code it needs in place on the current alpha, just a bit of tweaking is required. :)

dennyrgood, 2019-08-21:
I played with this a bit ... when I add tree supports to a model with the SL1 profile, save it with tree supports and bring it back in with it set to the FDM printer ... I can see the trees, but when I go to slice it the tops of the trees do not appear to touch/or come very close to the object.

I have not tried printing it yet, but has anyone else seen what I am referencing?

Mr Warren Schultz (and even others) has this been an issue for you?

Thanks!!!

tonypartridge, 2019-09-30:
+1 please!

stuartpb, 2020-01-31:
> The SLA tree supports will certainly not work for FDM without major changes.

What major changes would those be?

Kiron21, 2020-02-19:
+10

Ugars, 2020-03-20:
+1
Tree support is my only reason for using Cura.

Kiron21, 2020-03-20:
And ironing.

lars-sorensen, 2020-03-23:
Oh yes please!!

Nandox7, 2020-03-24:
> +1 Tree supports for FDM would be fantastic.
> Code is already there for SLA, so how hard can it be?
> 
> Also, "painting" an area on an object that needs support, with supports being auto-generated to target that specific area. That would be really useful.

Maybe create a second FR for that.
That one of feature I miss form cura, where it highlights the overhang areas and we can go around "clicking" on the ones we want support (or the reverse, where you don't want)
On more complex parts it make the work of adding soo much easier and fast compared to creating one support enforcer for every single detail.

Bytewaste, 2020-08-30:
+1 Would love some tree support, I fall back to Cura atm for that, but I like slic3r better overall.

jup2000, 2020-09-10:
+1 Please add tree support!  One of the few reasons I still go back to Cura is for their tree support.  It's second to none.

trollhunterD, 2020-10-24:
+1000
I been using the SL1 profile to then exporting the file with supports and reloading it to prusa under mk3 just to print wish we could just do it  the support option

cnlson, 2020-11-02:
@bubnikv please tell me which looks better? 

[image]
or 
[image]

especially when the supports come off so much easier
[image]

people are already using sl1 supports. why not make it easier for them if they want it?

Kampe, 2020-11-08:
gonna have to also +1 this one

MikeRatcliffe, 2020-11-09:
+1 just for the heck of it... the painted supports are awesome and I am sure the staff all know we all want tree supports.

You can always add them using meshmixer until Prusa implement it though ;)

karambolis8, 2020-11-17:
Waiting for this as others

sej7278, 2020-11-19:
just tried 2.3.0 alpha3 and was surprised to find no conical/treelike supports or fuzzy skin, we do seem to have ironing though

sej7278, 2020-11-29:
Let's face it, linear supports are unusable, most of the time they either snap the very model you're trying to support during removal or use too much filament and need too much cleanup. 

Treelike supports in other slicers aren't great either as they can be too fragile, so what's the middle ground - or is there another option?

cnlson, 2020-11-29:
@sej7278 if you look at the pictures from my post, that is using the resin printer supports, exported with supports, Then reimported as fdm and sliced. It works, just not totally optimized for FDM. They could do it, but in the meantime you can try it easily. the other option is meshmixer, but prusaslicer's print turned out much better than meshmixers version.

dryphi, 2020-12-01:
I like the PrusaSlicer tree supports better than MeshMixer because they place tree tips in between (i.e. at top and bottom surfaces of) the model instead of just the overhangs (image attached). Also it's much simpler to use just click and go. However the lack of options is a double-edged sword. I would like to be able to adjust some settings directly from the dialog. Also I'd like to print directly on the print-bed which would be useful even for SLA prints. I get the following error when I try removing the object elevation.
[image]
[image]

AmyTheCute, 2020-12-06:
this would be simply amazing. even for a dual extruder printer sometimes the support material doesn't adhere well to the other material like petg/pla and it can be risky to have it print on top of PETG because the slightest thing could make it fall off

Wright4TheJob, 2020-12-18:
+1 I often consider jumping over to cura purely for their tree support implementation, but my profiles are all nicely tuned in PrusaSlicer.

DanielLaberge, 2020-12-25:
Tree supports is my biggest wish for PrusaSlic3r and the only reason I still have CURA installed.

wtfiAndrewNelson, 2020-12-31:
Cura-like tree supports would be the most important new feature

veggysaurus, 2021-01-01:
+1 for tree supports
Do we really still have to use Cura or use SLA function, export with tree then import.

Battlevaria, 2021-01-01:
+1. Printing miniatures for gaming would be so much easier with this (or at least #4528). Right now I have to look for only models designed to be support-free, as the vertical-only supports have been disappointing again and again.

mneuhaus, 2021-01-05:
Would also love Something like this to better print thin tall Pieces with some "scaffolds" to keep it more stable

RaizerZolk, 2021-01-21:
+1 tree supports in Prusa Slicer for FDM prints.

mz83, 2021-01-23:
+1

Would prefer real tree branches for FDM, like in Cura though (much less susceptible to breaking than thin SLA-style trees).

dexter323i, 2021-01-25:
Yes, implement tree supports please! It would be the last step to totally beat Cura. (For me at least.)

sarvenn, 2021-01-26:
> Yes, implement tree supports please! It would be the last step to totally beat Cura. (For me at least.)

Fuzzy skin, horizontal hole expansion, wire printing, mould feature, conical support is also missing in PS.

Sleslie706, 2021-01-29:
Tree supports, yes please!

thatguyyoulove, 2021-01-31:
Tree supports are really beneficial when using some of the hotter warp-prone thermoplastics that can't utilize fan cooling as well to create bottom interfaces that easily separate from the model. By having the support snake around objects, holes, etc. to the build plate without having to build off of printed surfaces, you can avoid this issue entirely which I've found extremely useful with ABS, Nylon, and Polycarbonate in Cura. Would love to be able to take advantage of this in Prusaslicer and use it as my sole slicer moving forward.

melophat, 2021-02-01:
2 years and obviously a ton of requests for tree supports and still no dice..  please get this done in the 2.4 release!

tckb, 2021-02-16:
Does anyone have a workaround or suggestion on how to do this in Prusa? meshmixer support import gets the  dimension wrong. I need this to support open box.  
[image]

martinowren, 2021-02-16:
> Does anyone have a workaround or suggestion on how to do this in Prusa? meshmixer support import gets the dimension wrong. I need this to support open box.
> [image]

Wrong place to post ;) Make a unique pull-request or search for Prusa3D groups on facebook.

tckb, 2021-02-16:
> > Does anyone have a workaround or suggestion on how to do this in Prusa? meshmixer support import gets the dimension wrong. I need this to support open box.
> 
> Wrong place to post ;) Make a unique pull-request or search for Prusa3D groups on facebook.

Probably not, the issue I have is totally solved by using tree support. Another Issue for the same problem makes no sense and will most likely be buried in the plethora of unanswered Issues. Thought a question and an image is much better than a meager  "+1"

martinowren, 2021-02-16:
> Probably not, the issue I have is totally solved by using tree support. Another Issue for the same problem makes no sense and will most likely be buried in the plethora of unanswered Issues. Thought [...]

This is a PR to get tree support implemented, not help with tree support or prusaslicer ;)

tckb, 2021-02-16:
> This is a [feature request for a] PR to get tree support implemented

As a non-native speaker, this is sufficiently simple to understand that. 

>not help with tree support or prusaslicer ;)

based on 2+ years of no-activity and sufficient interest, I sought help from the community (also I did searched elsewhere) if there were any possible workarounds. Lets not get carried away here. Unless, there's anything useful here, I would end conversation here to prevent spamming the subscribers. cheers!

viestursr, 2021-02-19:
This is the most thumbs-upped and 2nd most commented open issue. Please make this happen, it's been 2 years.

Ibbanez, 2021-03-01:
+1 Please

PapalPotato, 2021-03-07:
+1
Hope this makes it in at least an alpha soon

Syco54645, 2021-03-08:
+1 here. certain things will keep me grounded in cura. this is one of them.

DavidUnboxed, 2021-03-12:
+1 
Would like to see tree supports for i3 MKS3+MMU2S with option to use soluble (PVA) for supports and support interfaces.

Ali229, 2021-03-12:
Need this really important feature on Prusa +1

DrPippin, 2021-03-17:
Tree supports are definitely a request of mine, too.

clfaye, 2021-03-20:
+1 Please add tree support

marcottt, 2021-03-28:
+1 old request but still useful...

nirurin, 2021-04-01:
Many years later and still no tree supports? Is it even on the roadmap?

CobaltEcho, 2021-04-01:
> Many years later and still no tree supports? Is it even on the roadmap?

@nirurin Wait, there is a roadmap?!

CobaltEcho, 2021-04-01:
@bubnikv Would it help to get any other labels added to this request/feature/issue so it could be looked at?  Or any updates provided?  In addition, as a community, is there anything we can do to assist with this request/feature/issue?

nirurin, 2021-04-01:
I may actually have to move back to cura, as I've realised that tree supports basically seem to be the superior option. PrusaSlicer is too many years behind the curve.

[70 more replies]

closed 2023-02-01 by lukasmatena