prusa3d/PrusaSlicer #1515: make install doesn't install anything [1.42.0-alpha1]

Moredread, 2018-12-25:
### Version

1.42.0-alpha1

### Operating system type + version
Linux NixOS 18.09

### Behavior

I'm currently packaging 1.42.0-alpha1 for NixOS.

`make install` doesn't install anything anymore. Is this intended?

bubnikv, 2018-12-25:
The make install is not finished, it was not even initiated.

smellyfis, 2019-01-02:
I tried to fix this with pull request #1553

bubnikv, 2019-01-02:
We were discussing the install with @vojtechkral . We are considering to install a shell script into the bin directory and to call the proper Slic3r binary from there, providing it with path to the resources directory. This way we would not have to recompile Slic3r to be executed from the system or from Appimage or for debugging without installation.

smellyfis, 2019-01-02:
I feel that a wrapper script is a slippery slope to bad programming practices so be careful. 

Bad practice would be relying on an envirnoment variable eg. SLIC3R_APP

good practice would be to read a set of configuration file eg for linux /etc/slic3r-prusa3d/config.ini or ~/.slic3r-prusa3d/config.ini (or both)

bubnikv, 2019-01-02:
Reading a configuration file from a compiled-in path does not solve our
issue, it just wraps it into a new coat.

IMHO it makes sense to add a command line parameter, something like
--resourcesdir and set this parameter from the script. This goes hand in
hand with the --datadir parameter we already have.

vojtechkral, 2019-01-03:
@smellyfis This isn't configuration, this is application resources. Configuration is a separate issue, cf. #1526.

As for your pull request, it cannot be merged the way it is right now, because `RESOURCE_PATH` is set on all Unix OSes in a blanket way, which would break our AppImage and Mac OS builds.

I can see two solutions. One is to have an explicit CMake option to tell Slic3r that's it's installed in the 'traditional' UNIX way and where to look for resources. The other is installing Slic3r's whole directory structure into `/usr/lib/slic3r-prusa3d` and placing a trampoline script in `/usr/bin`. This is what Firefox does, for example, as well as some other software I believe.

We'll think about this and I'd also like to ask @Moredread and @hyperair for an opinion, since they actually need this to produce packages.

smellyfis, 2019-01-03:
Understood about messing with Mac and appimage. What about on the setting of resource path we check if install configuration is not "appimage"? (As oppose to "Release" or "Debug") that way you could set you up to have different compilerflags even for appimage or other Unix like systems.

The other option I see is best is make an option flag that hides the resource_path unless it is turned on then proceeds accordingly.

I think installing all everything into one directory is not very good as it doesn't follow the logic of the FHS?  Not following it makes it a nightmare for package maintainers for different distros to have to figure out how to install properly.

I can modify my pull request #1553 to do either of these options if y'all would like.

vojtechkral, 2019-01-03:
> I think installing all everything into one directory is not very good as it doesn't follow the logic of the FHS? Not following it makes it a nightmare for package maintainers for different distros to have to figure out how to install properly.

That's why I paged the two pkg maintainers that I know of for an opinion. Let's wait and see if any of them replies whether they need a strict adherence to the FHS or wether installing into one dir "Firefox style" would be good enough.

hyperair, 2019-01-03:
In Debian, we do adhere to the FHS as described at https://www.debian.org/doc/packaging-manuals/fhs/fhs-2.3.html. Here's the [file list of the slic3r-prusa package in Debian](https://packages.debian.org/sid/amd64/slic3r-prusa/filelist), and the [list of patches](https://salsa.debian.org/3dprinting-team/slic3r-prusa/tree/master/debian/patches) that are applied during the package building process to make this happen.

I'm not sure if installing everything into one directory is acceptable, but it certainly isn't recommended practice and I'm not keen on adopting it.

TL;DR: resources go into /usr/share/slic3r-prusa3d, and code files go into /usr/lib/slic3r-prusa3d.

smellyfis, 2019-01-03:
I made changes to the pull request #1553 to check the CMAKE_BUILD_TYPE as well as if it not APPLE.

https://github.com/smellyfis/Slic3r/blob/4b45713be9fb7e45fa74a35d225a08cbdb7fd55d/CMakeLists.txt#L282-L285

Moredread, 2019-01-03:
On NixOS each application is installed to it's separate folder, so a configurable option e.g. via cmake would be great. :)

smellyfis, 2019-01-03:
I made an option SLIC3R_APPIMAGE that when set (default) then it ignores setting RESOURCE_PATH to the DATADIR  ie skipping the clause I put into slic3r.cpp
Also followed the same logic for installing resources

to turn off Appimage and install in the FHS on a *NIX system that isn't APPLE run with
```bash
cmake -DSLIC3r_APPIMAGE=no -DCMAKE_INSTALL_PREFIX=/usr 
```

hyperair, 2019-01-04:
I had a brief look at the [AppImage packaging documentation for cmake projects](https://docs.appimage.org/packaging-guide/from-source/native-binaries.html#cmake), and you don't need to do anything special with it. I don't see any need for throwing everything into one folder for AppImages.

bubnikv, 2019-01-04:
What method do you recommend to direct Slic3r to pull its resources from
the source directory or from the standalone binary .tar.gz package for the
purpose of development and testing isolated from the system installation? I
personally do not like to hard code the resources path into the slic3r
binary based on the purpose (install to system, install to binary .tar.gz,
or run it isolated for development). But then adding a command line switch
like --resourcesdir to override the hard coded path would lead to
confusion. For example, this would lead to github issues asking, why Slic3r
does not start from the source directory, or why it is behaving
incorrectly, as it would pull the resources from the system installation by
default. Any thoughts?

I personally like the solution, where the path is hard coded relative to
the binary, and one would install a wrapper script into system.

vojtechkral, 2019-01-04:
@hyperair Thanks for replying here. As for the set of patches, the ones related to Perl won't be necessary in the next Slic3r PE version since the installed Slic3r doesn't need Perl. As for the other ones, I'd like to fix Slic3r such that they aren't necessary.
As for the resource path, we would like to keep the default of binary-relative lookup (this is desirable for developement/testing and non-Linux platforms) but I think we can provide a CMake option (perhaps something like `SLIC3R_FHS` or the like) that you could use which would configure the path according to the FHS. Would that be ok?

smellyfis, 2019-01-04:
@vojtechkral Why not use something like the system path for the resources? So it first looks at a user-supplied run-time specified location then a compile-time location if it doesn't exist in either place it looks in a binary-relative location? Then You can have the install location for your testing be ${CMAKE_BINARY_DIR} or ${CMAKE_CURRENT_BINARY_DIR} for target platforms that need it to be such. Or even just use the ${CMAKE_INSTALL_PREFIX} as it is intended and base the compile-time locations relative to that.

vojtechkral, 2019-01-04:
@smellyfis Because any of that constitutes quite a lot of additional complexity of setup needed when building, and/or behaviour of the resulting program while at the same time I don't see really any actual problem being solved. Right now I'd like to ease the job for pkg maintainers with preferably low-impact changes that don't break anything else.

I mean sure, the way Slic3r is built and installed is far from perfect, but there's only so many hours to a workday and the backlog of other issues is substantial.

vojtechkral, 2019-01-07:
@hyperair The install targets should be fixed now and I've added the `SLIC3R_FHS` option which makes Slic3r look for the appropriate `share/slic3r-prusa3d` directory. Hopefully that's ok Debian-wise? Cf. also the explanation in `doc`. The one remaining issue is the expat lookup, hopefully I'll have a look at that at some point as well. Later on when the final release is out and if/when you get around to packaging it, feel free to reach out to me if there are issues.

ArchangeGabriel, 2019-01-11:
For the record, I will be packaging this project in ArchLinux repo when 1.42 final will be released, but for what matters here, the basic principle is that whatever suits Debian generally suits us as well. ;)

I’m not a CMake expert at all so I don’t know if there is an easier way than having to rely on `SLIC3R_FHS`, but I’m fine with it anyway. :)

vojtechkral, 2019-01-12:
@ArchangeGabriel Cool. Will you be packaging for the AUR or one of the "proper" repos? (`community`?)

mkyral, 2019-01-12:
Hi,
I just compilled and installed latest git version. All OK, but on start I get this error:

```
ASSERT INFO:
/var/tmp/portage/media-gfx/slic3r-pe-9999/work/slic3r-pe-9999/src/slic3r/GUI/GUI_App.cpp(89): assert "wxDirExists(resources_dir)" failed in OnInit(): Resources path does not exist or is not a directory: /home/marian/../resources

BACKTRACE:
[1] Slic3r::GUI::GUI_App::OnInit()
[2] wxEntry(int&, wchar_t**)
[3] main
[4] __libc_start_main
[5] _start
```

**Installed files:**
```
/usr/share/slic3r-prusa3d/
/usr/share/slic3r-prusa3d/shaders/
/usr/share/slic3r-prusa3d/shaders/variable_layer_height.vs
/usr/share/slic3r-prusa3d/shaders/variable_layer_height.fs
/usr/share/slic3r-prusa3d/shaders/gouraud.vs
/usr/share/slic3r-prusa3d/shaders/gouraud.fs
/usr/share/slic3r-prusa3d/profiles/
/usr/share/slic3r-prusa3d/profiles/PrusaResearch.ini
/usr/share/slic3r-prusa3d/profiles/PrusaResearch.idx
/usr/share/slic3r-prusa3d/models/
/usr/share/slic3r-prusa3d/models/mk3_bed.stl
/usr/share/slic3r-prusa3d/models/mk2_bed.stl
/usr/share/slic3r-prusa3d/localization/
/usr/share/slic3r-prusa3d/localization/zh_CN/
/usr/share/slic3r-prusa3d/localization/zh_CN/Slic3rPE_zh.po
/usr/share/slic3r-prusa3d/localization/zh_CN/Slic3rPE.mo
/usr/share/slic3r-prusa3d/localization/uk/
/usr/share/slic3r-prusa3d/localization/uk/Slic3rPE_uk.po
/usr/share/slic3r-prusa3d/localization/uk/Slic3rPE_.mo
/usr/share/slic3r-prusa3d/localization/uk/Slic3rPE.mo
/usr/share/slic3r-prusa3d/localization/pl/
/usr/share/slic3r-prusa3d/localization/pl/Slic3rPE_pl.po
/usr/share/slic3r-prusa3d/localization/pl/Slic3rPE.mo
/usr/share/slic3r-prusa3d/localization/list.txt
/usr/share/slic3r-prusa3d/localization/ko_KR/
/usr/share/slic3r-prusa3d/localization/ko_KR/Slic3rPE.po
/usr/share/slic3r-prusa3d/localization/ko_KR/Slic3rPE.mo
/usr/share/slic3r-prusa3d/localization/it/
/usr/share/slic3r-prusa3d/localization/it/Slic3rPE_it.po
/usr/share/slic3r-prusa3d/localization/it/Slic3rPE.mo
/usr/share/slic3r-prusa3d/localization/fr_FR/
/usr/share/slic3r-prusa3d/localization/fr_FR/Slic3rPE_fr.po
/usr/share/slic3r-prusa3d/localization/fr_FR/Slic3rPE.mo
/usr/share/slic3r-prusa3d/localization/es/
/usr/share/slic3r-prusa3d/localization/es/Slic3rPE_es.po
/usr/share/slic3r-prusa3d/localization/es/Slic3rPE.mo
/usr/share/slic3r-prusa3d/localization/en_US/
/usr/share/slic3r-prusa3d/localization/en_US/Slic3rPE_en.po
/usr/share/slic3r-prusa3d/localization/en_US/Slic3rPE.mo
/usr/share/slic3r-prusa3d/localization/de_DE/
/usr/share/slic3r-prusa3d/localization/de_DE/Slic3rPE_de.po
/usr/share/slic3r-prusa3d/localization/de_DE/Slic3rPE.mo
/usr/share/slic3r-prusa3d/localization/cs_CZ/
/usr/share/slic3r-prusa3d/localization/cs_CZ/Slic3rPE_cs.po
/usr/share/slic3r-prusa3d/localization/cs_CZ/Slic3rPE.mo
/usr/share/slic3r-prusa3d/localization/Slic3rPE.pot
/usr/share/slic3r-prusa3d/icons/
/usr/share/slic3r-prusa3d/icons/zoom.png
/usr/share/slic3r-prusa3d/icons/wrench.png
/usr/share/slic3r-prusa3d/icons/wand.png
/usr/share/slic3r-prusa3d/icons/view_toolbar.png
/usr/share/slic3r-prusa3d/icons/variable_layer_height_tooltip.png
/usr/share/slic3r-prusa3d/icons/variable_layer_height_reset.png
[... 151 lines cut ...]
/usr/share/slic3r-prusa3d/icons/application_view_tile.png
/usr/share/slic3r-prusa3d/icons/add_object.png
/usr/share/slic3r-prusa3d/icons/add.png
/usr/share/slic3r-prusa3d/icons/action_undo_grey.png
/usr/share/slic3r-prusa3d/icons/action_undo.png
/usr/share/slic3r-prusa3d/icons/Slic3r_32px.png
/usr/share/slic3r-prusa3d/icons/Slic3r_192px_transparent.png
/usr/share/slic3r-prusa3d/icons/Slic3r_192px_grayscale.png
/usr/share/slic3r-prusa3d/icons/Slic3r_192px.png
/usr/share/slic3r-prusa3d/icons/Slic3r_128px.png
/usr/share/slic3r-prusa3d/icons/Slic3r.png
/usr/share/slic3r-prusa3d/icons/Slic3r.ico
/usr/share/slic3r-prusa3d/icons/Slic3r.icns
/usr/share/slic3r-prusa3d/icons/Slic3r-console.ico
/usr/share/slic3r-prusa3d/fonts/
/usr/share/slic3r-prusa3d/fonts/NotoSans-hinted.zip
/usr/share/slic3r-prusa3d/fonts/NotoSans-Regular.ttf
/usr/share/slic3r-prusa3d/avrdude/
/usr/share/slic3r-prusa3d/avrdude/avrdude.conf
/usr/share/doc/slic3r-pe-9999/
/usr/share/doc/slic3r-pe-9999/README.md.bz2
/usr/bin/slic3r-gui
/usr/lib/debug/usr/bin/slic3r-gui.debug
```
Is it bug or I missed something?

mkyral, 2019-01-12:
OK,
fixed by adding `-DSLIC3R_FHS=ON`

vojtechkral, 2019-01-13:
@mkyral Glad you got it working.

@ArchangeGabriel That's cool to hear. I'm a long-time Arch user myself, but since I do Slic3r dev I don't really have a need for a package :)

Moredread, 2019-01-28:
@vojtechkral It works with NixOS. :)

I'm going to package the stable version as soon as it is available. In the meantime, if someone using Nix is reading this, I have a derivation available for the latest version in https://github.com/Moredread/nur-packages

commit 8bc8dfab (2019-01-07): Build: Fix install targets, add SLIC3R_FHS, check resource dir on GUI_App init
Fix #1515