prusa3d/PrusaSlicer #1516: Error message in case of a gcode error is too small [1.42.0-alpha1]

Boby71, 2018-12-25:
### Version
1.42.0-alpha1

### Operating system type + version
Win64

### Behavior
I had an error in my gcode starting script and was using "send to printer" and was wondering, why nothing arrived at my octoprint instance. After a few minutes, I noticed a small error message in the bottom of the window telling me, the gcode generation has failed.

My expectation: If there is an error which prevents the software from generating a working gcode, at least a box should appear showing this message (if not modal).

[image]

bubnikv, 2018-12-25:
Please as usual, attach your 3MF.

Boby71, 2018-12-25:
Here you are:
[retraction_test_v1-1.zip](https://github.com/prusa3d/Slic3r/files/2708980/retraction_test_v1-1.zip)
Thanks,
Boby

bubnikv, 2019-01-02:
The problem is in the "degree" sign you placed into your custom start G-code.

The error message guides you to inspect the temp G-code and to search for the error message. This is what I have found:

!!!!! Failed to process the custom G-code template start_gcode
Parsing error at line 2: Invalid utf8 sequence
M301 P24.40 I2.12 D75.95; set PID for extruder heater @ 190°C                                                           ^
!!!!! End of an error report for the custom G-code template start_gcode

bubnikv, 2020-12-03:
I have improved error reporting of errors in output file name templates with 306acbdfb48ae5ded9772605f6b3a2d6ca911d3f
The errors are now displayed in a mono-spaced font, so that the arrow character ^ is displayed at the right column
pointing to the offending spot.
[image]

bubnikv, 2020-12-03:
Further improved with 3dd9b8c718c22e4ee7a000341b6d6c72b23df206

    Further improvement error reporting with buggy custom G-code sections #1516
    1) The macro-processor sanitizes the source code line for invalid UTF-8
       characters. Ideally these invalid characters would be replaced with ?,
       they are just dropped as of now. Better than showing an empty string
       when converting to wxString though.
    2) G-code export collects full error message for 1st occurence of an error
       for each custom G-code block.
    3) The composite error message now displays the errors collected in 2).
    4) The error window is now scaled bigger and the Slicer logo is smaller
       if the text is to be rendered with monospaced font, as the monospaced
       text will not be word wrapped.

Actually your original issue was that you entered a non-UTF8 character into the custom G-code section.

[image]

bubnikv, 2020-12-03:
The non-conforming UTF8 character was removed from the message, but at least the offending line is there and the arrow points to the location from which the offending character was removed and an "Invalid utf8 sequence" hint is displayed. IMHO this behavior is quite intuitive.

It will be in PrusaSlicer 2.3.0-beta2
Closing.

closed 2020-12-03 by bubnikv